Ysurac/openmptcprouter #2765: Connection problem some devices "DHCP"

jccourtoy, 2023-02-05:
I have a problem connecting some devices. Example on my daughter's smartphone, it is connected "without internet" but with the web accessible. And, with exactly the same connection settings on my smartphone, it works.
I put the phones in fixed IP, configure other DNS, reinstall the server. Add DNS in OpenMTPC
I do not know what to do
[image]

## Specifications

  - OpenMPTCProuter version: 5.4.207-mptcp 0.1028
  - OpenMPTCProuter VPS version: 0.1028 5.4.207-mptcp
  - OpenMPTCProuter VPS provider: Onetsolution
  - OpenMPTCProuter platform: Raspberry Pi 4 Model B Rev 1.4

Sys log 
Sun Feb  5 14:03:38 2023 user.notice omr-tracker: Launched
Sun Feb  5 14:03:38 2023 user.notice omr-bypass: Starting OMR-ByPass...
Sun Feb  5 14:03:39 2023 daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(eth0) 192.168.66.208 00:04:4b:e4:b2:04
Sun Feb  5 14:03:39 2023 daemon.info dnsmasq-dhcp[1]: DHCPACK(eth0) 192.168.66.208 00:04:4b:e4:b2:04
Sun Feb  5 14:03:39 2023 daemon.info omr-tracker-v2ray: V2Ray is up (can contact via http 1.0.0.1)
Sun Feb  5 14:03:40 2023 user.notice omr-bypass: OMR-ByPass is running
Sun Feb  5 14:03:40 2023 daemon.info glorytun: starting glorytun vpn instance vpn
Sun Feb  5 14:04:00 2023 user.notice SQM-autorate: Launch on wan1
Sun Feb  5 14:04:00 2023 user.notice SQM-autorate: Launch on wan2
Sun Feb  5 14:04:02 2023 user.notice SQM-autorate: Launch on omrvpn
Sun Feb  5 14:04:02 2023 user.notice SQM-autorate: Launch on lan
Sun Feb  5 14:05:12 2023 user.notice SQM-autorate: Launch on wan1
Sun Feb  5 14:05:12 2023 user.notice SQM-autorate: Launch on wan2
Sun Feb  5 14:05:14 2023 user.notice SQM-autorate: Launch on omrvpn
Sun Feb  5 14:05:14 2023 user.notice SQM-autorate: Launch on lan
Sun Feb  5 14:06:24 2023 user.notice SQM-autorate: Launch on wan1
Sun Feb  5 14:06:24 2023 user.notice SQM-autorate: Launch on wan2
Sun Feb  5 14:06:26 2023 user.notice SQM-autorate: Launch on omrvpn
Sun Feb  5 14:06:26 2023 user.notice SQM-autorate: Launch on lan
Sun Feb  5 14:07:37 2023 user.notice SQM-autorate: Launch on wan1
Sun Feb  5 14:07:37 2023 user.notice SQM-autorate: Launch on wan2
Sun Feb  5 14:07:39 2023 user.notice SQM-autorate: Launch on omrvpn
Sun Feb  5 14:07:39 2023 user.notice SQM-autorate: Launch on lan
Sun Feb  5 14:07:42 2023 user.notice OMR-VPS: Get config from server vps...
Sun Feb  5 14:07:42 2023 user.notice OMR-VPS: V2ray restart...
Sun Feb  5 14:07:43 2023 daemon.info v2ray: DNS disabled: main_dns
Sun Feb  5 14:07:43 2023 daemon.info v2ray: Setting transparent proxy on port: 1897
Sun Feb  5 14:07:43 2023 daemon.info v2ray: Transparent proxy mode: default
Sun Feb  5 14:07:43 2023 user.notice v2ray: add rules
Sun Feb  5 14:07:43 2023 user.notice v2ray: v2ray-rules -l 1897 -L 1897 -s 185.163.124.224 --rule-name def --src-default forward --dst-default forward --local-default forward
Sun Feb  5 14:07:43 2023 user.notice v2ray: Reload omr-bypass rules
Sun Feb  5 14:07:43 2023 user.notice omr-bypass: Starting OMR-ByPass...
Sun Feb  5 14:07:43 2023 daemon.info glorytun[13342]: STOPPED tun0
Sun Feb  5 14:07:43 2023 daemon.notice netifd: Network device 'tun0' link is down
Sun Feb  5 14:07:43 2023 daemon.notice netifd: Interface 'omrvpn' has link connectivity loss
Sun Feb  5 14:07:43 2023 daemon.notice netifd: Interface 'omrvpn' is now down
Sun Feb  5 14:07:43 2023 daemon.info glorytun: starting glorytun vpn instance vpn
Sun Feb  5 14:07:43 2023 daemon.info glorytun[2200]: INITIALIZED tun0
Sun Feb  5 14:07:43 2023 daemon.notice netifd: Interface 'omrvpn' is disabled
Sun Feb  5 14:07:43 2023 daemon.notice netifd: Interface 'omrvpn' is enabled
Sun Feb  5 14:07:43 2023 daemon.notice netifd: Interface 'omrvpn' is disabled
Sun Feb  5 14:07:43 2023 daemon.notice netifd: Interface 'omrvpn' is enabled
Sun Feb  5 14:07:43 2023 daemon.notice netifd: Network device 'tun0' link is up
Sun Feb  5 14:07:43 2023 daemon.notice netifd: Interface 'omrvpn' has link connectivity
Sun Feb  5 14:07:43 2023 daemon.notice netifd: Interface 'omrvpn' is setting up now
Sun Feb  5 14:07:43 2023 daemon.notice netifd: Interface 'omrvpn' is now up
Sun Feb  5 14:07:44 2023 user.notice omr-tracker: Launching...
Sun Feb  5 14:07:44 2023 daemon.err glorytun[2200]: 185.163.124.224.65001: connected
Sun Feb  5 14:07:44 2023 daemon.info glorytun[2200]: STARTED tun0
Sun Feb  5 14:07:44 2023 daemon.warn [5617]: <warn>  [SQM_IFB_aed76] invalid sysfs path read for net/SQM_IFB_aed76
Sun Feb  5 14:07:44 2023 daemon.warn [5617]: <warn>  [base-manager] couldn't handle kernel event: device net/SQM_IFB_aed76 not found
Sun Feb  5 14:07:44 2023 daemon.warn [5617]: <warn>  [SQM_IFB_f6184] invalid sysfs path read for net/SQM_IFB_f6184
Sun Feb  5 14:07:44 2023 daemon.warn [5617]: <warn>  [base-manager] couldn't handle kernel event: device net/SQM_IFB_f6184 not found
Sun Feb  5 14:07:44 2023 user.notice post-tracking-post-tracking: Set firewall on server vps
Sun Feb  5 14:07:45 2023 daemon.warn [5617]: <warn>  [SQM_IFB_03ea1] invalid sysfs path read for net/SQM_IFB_03ea1
Sun Feb  5 14:07:45 2023 daemon.warn [5617]: <warn>  [base-manager] couldn't handle kernel event: device net/SQM_IFB_03ea1 not found
Sun Feb  5 14:07:45 2023 daemon.warn [5617]: <warn>  [SQM_IFB_01fd3] invalid sysfs path read for net/SQM_IFB_01fd3
[... 339 lines cut ...]
Sun Feb  5 15:54:23 2023 daemon.notice netifd: Interface 'lan' is setting up now
Sun Feb  5 15:54:23 2023 daemon.notice netifd: Interface 'lan' is now up
Sun Feb  5 15:54:23 2023 daemon.notice netifd: Network device 'eth0' link is up
Sun Feb  5 15:54:23 2023 daemon.notice netifd: Interface 'lan' has link connectivity
Sun Feb  5 15:54:23 2023 kern.info kernel: [ 6839.516702] bcmgenet fd580000.ethernet eth0: Link is Up - 100Mbps/Full - flow control rx/tx
Sun Feb  5 15:54:23 2023 kern.info kernel: [ 6839.525403] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
Sun Feb  5 15:54:23 2023 user.notice firewall: Reloading firewall due to ifup of lan (eth0)
Sun Feb  5 15:54:23 2023 user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Sun Feb  5 15:54:24 2023 user.notice mptcp: Reloading mptcp config due to ifup of lan (eth0)
Sun Feb  5 15:54:24 2023 user.notice post-tracking-post-tracking: Set firewall on server vps
Sun Feb  5 15:54:25 2023 daemon.info dnsmasq[1]: read /etc/hosts - 4 addresses
Sun Feb  5 15:54:25 2023 daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 2 addresses
Sun Feb  5 15:54:25 2023 daemon.info dnsmasq-dhcp[1]: read /etc/ethers - 0 addresses
Sun Feb  5 15:54:28 2023 user.notice omr-bypass: Starting OMR-ByPass...
Sun Feb  5 15:54:29 2023 user.notice omr-bypass: OMR-ByPass is running
Sun Feb  5 15:54:29 2023 user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Sun Feb  5 15:54:30 2023 user.notice post-tracking-post-tracking: Set firewall on server vps
Sun Feb  5 15:59:04 2023 daemon.info dnsmasq-dhcp[1]: DHCPDISCOVER(eth0) 82:b7:90:bb:23:c1
Sun Feb  5 15:59:04 2023 daemon.info dnsmasq-dhcp[1]: DHCPOFFER(eth0) 192.168.66.210 82:b7:90:bb:23:c1
Sun Feb  5 15:59:04 2023 daemon.info dnsmasq-dhcp[1]: DHCPDISCOVER(eth0) 82:b7:90:bb:23:c1
Sun Feb  5 15:59:04 2023 daemon.info dnsmasq-dhcp[1]: DHCPOFFER(eth0) 192.168.66.210 82:b7:90:bb:23:c1
Sun Feb  5 15:59:04 2023 daemon.info dnsmasq-dhcp[1]: DHCPDISCOVER(eth0) 82:b7:90:bb:23:c1
Sun Feb  5 15:59:04 2023 daemon.info dnsmasq-dhcp[1]: DHCPOFFER(eth0) 192.168.66.210 82:b7:90:bb:23:c1
Sun Feb  5 15:59:04 2023 daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(eth0) 192.168.66.210 82:b7:90:bb:23:c1
Sun Feb  5 15:59:04 2023 daemon.info dnsmasq-dhcp[1]: DHCPACK(eth0) 192.168.66.210 82:b7:90:bb:23:c1 Note10-de-Satine

Ysurac, 2023-02-05:
I don't really understand what you mean by "without internet but with the web accessible". You have a message that internet is not working but in fact it is ? Did you try so see internet ip of the phone using ifconfig.co to check if it use really the VPS ?

Network-Traditions, 2023-03-06:
Most device OSes "phone home" to validate the "internet" connectivity.  This includes Windows on desktops, Android on smartphones, etc.  If the device is unable to complete the "phone home" process, it will report NO "internet", but the device may be able to freely access many if not all its typical internet sources.  For our pfsense/OMR configuration, we get the same notice on our Android smartphones when there are connection issues with OMR, the VPS and any transitional operations such as OMR/VPS reboots, reconnects, etc.  If the status remains as indicated, something (firewall, service provider, etc) is likely blocking the "phone home" process for your device.

closed 2023-06-10 by github-actions[bot]