Ysurac/openmptcprouter #2766: Installing `luci-ssl` to provide SSL for web-interface fails

paulkefer, 2023-02-06:
## Expected Behavior
```
opkg update
opkg install luci-ssl
/etc/init.d/uhttpd restart
```

luci should now be served with SSL

## Current Behavior
```
opkg update
opkg install luci-ssl
Installing luci-ssl (git-20.244.36115-e10f954) to root...
Downloading https://packages.openmptcprouter.com/v0.59.1-5.4/aarch64_cortex-a72/luci/luci-ssl_git-20.244.36115-e10f954_all.ipk
Installing libustream-wolfssl20201210 (2022-01-16-868fd881-1) to root...
Downloading https://packages.openmptcprouter.com/v0.59.1-5.4/aarch64_cortex-a72/base/libustream-wolfssl20201210_2022-01-16-868fd881-1_aarch64_cortex-a72.ipk
Collected errors:
 * check_data_file_clashes: Package libustream-wolfssl20201210 wants to install file /lib/libustream-ssl.so
	But that file is already provided by package  * libustream-openssl20201210
 * opkg_install_cmd: Cannot install package luci-ssl.
```

## Specifications

  - OpenMPTCProuter version: 0.59.1-5.4
  - OpenMPTCProuter VPS version: 0.1028 5.4.207-mptcp
  - OpenMPTCProuter VPS provider: AWS
  - OpenMPTCProuter platform: RPI4B

Ysurac, 2023-02-06:
If you try https://192.168.100.1/ you should already use TLS. You can configure this via Services->uHTTPd.

paulkefer, 2023-02-06:
You're right - I mistook my browser's notification that it didn't know the certificate authority for its notification that HTTPS was not available; manually importing and trusting the certificate solved the issue. Thank you!

closed 2023-02-06 by paulkefer