Ysurac/openmptcprouter #2768: roundrobin don't work

ogmkp, 2023-02-07:
## Expected Behavior

Each packet should be sent across each WAN

## Current Behavior

In mptcp roundrobin, when I send a TCP session, it only goes through the master interface, there is no packet balancing.
It acts like BLEST or ECF with many TCP sessions open.
sysctl says roundrobin applied
WANs with different IPs with differents gateways are mounted over vlans ifaces binded on eth0
I checked everything from logs and tried all vpns and proxies from 1 week.

## Possible Solution

## Steps to Reproduce the Problem

  1. Send one session with iperf for example.
  2. Look at the master interface who takes all upload bandwidth but nothing on others.
  3. Galérer avec tous les paramètres sans réussir.

## Context (Environment)

## Specifications

  - OpenMPTCProuter version: openmptcprouter-v0.59.1-5.4-r0+16594-ce92de8c8c-x86-64-generic
  - OpenMPTCProuter VPS version: 0.59.1-5.4 r0
  - OpenMPTCProuter VPS provider: OVH
  - OpenMPTCProuter platform: x86_64
  - Country: France

Ysurac, 2023-02-08:
It's possible this doesn't work, it was not really tested and I don't think anyone use this.
You have a use case ?

ogmkp, 2023-02-08:
You can test by pushing an RTMP stream to youtube or a netcat or an iperf with a single ESTABLISHED TCP session.
The goal is to use the links to push each packet with loss control.
It seems that all path control fill the congestion buffer before using second link and only with new tcp sessions.
It works great with browsers because they use many tcp sessions for a http exchange but no with single threaded tcp.
Glorytun UDP same it only pushes through the master link, is it possible to remove the prio's on the routing table and have something equlibrated and have redundancy?
I don't know much where to look.

Ysurac, 2023-02-08:
round-robin doesn't provide any loss control, redundant mode should.
Connections are used based on latency, no way to change that for now.

ogmkp, 2023-02-08:
Is it possible in cli to change this mecanism to distribute over each wans ?

Ysurac, 2023-02-09:
redundant mode use all wans at same time.
Else, by default, wans are used based on latency and when needed.

ogmkp, 2023-02-09:
I saw open issues at mptcp upstream, hopefully they fix the issues and some of them are already fixed, I'm closing this ticket, there are others on this project asking the same thing. thanks for the work.

closed 2023-02-09 by ogmkp

ogmkp, 2023-02-13:
After a full reinstall of omr client and omr vps, the feature is working and BLEST reacts very better,
so one issue, when I enable roundrobin and reenable BLEST, the server doen't make switch back and stay on roundrobin, I have to reboot omr client.
Thank you @Ysurac for this project