Ysurac/openmptcprouter #2772: How to delay heartbeat checking time for 4G LTE Internets ?

gaamaaresosa, 2023-02-09:
## Expected Behavior
Set heart beat every 3000ms for poor 4G LTE internet latency to avoid server disconnection warnings.

## Current Behavior
4G LTE Huawei dongles used for internet.
On ideal state all the interfaces, tunnel and server are green color.
When a speed test is done, the ping latency increases more than 2000ms.
That time all goes to warning state.

How to delay heartbeat checking time for 4G LTE Internets ?

## Specifications

OpenMPTCProuter version: v0.59.1-5.4
OpenMPTCProuter VPS version: 5.4.207-mptcp
OpenMPTCProuter VPS provider: DigitalOcean
OpenMPTCProuter platform: RPI4 Kernal 5.4.194

### IDEAL STATE:
[image]

### SPEED TEST:
[image]

[image]

### POOR LATENCY WHILE SPEED TEST:
[image]

### TIME OUT WARNINGS:
[image]

Ysurac, 2023-02-09:
Use SQM with download and upload speed set.

gaamaaresosa, 2023-02-09:
> Use SQM with download and upload speed set.

I have already tried SQM, which is further reducing the speed.
In our country the 4G LTE upload speed is from 1Mbps to 4Mbps max

Ysurac, 2023-02-09:
Congestion in speedtest is not the same as disconnection due to week signal. In your post you are talking about congestion, and if SQM need to be set correctly but it's the only way to fix congestion as you have with speedtest.

For week signal disconnection, you can set TCP SYN retries in Network->MPTCP, scheduler to BLEST and play with congestion control. You can also set timeout in Services->OMR-Tracker.

OMR work on 4G, satellite, fiber,... If speedify give you a better result you should use it.

gaamaaresosa, 2023-02-10:
> Congestion in speedtest is not the same as disconnection due to week signal. In your post you are talking about congestion, and if SQM need to be set correctly but it's the only way to fix [...]

Thanks again Ysurac,
First of all I am fed up with 3 major mobile networks in our country.
All are hopeless, disconnects even while starting the speedtest.
Especially on Huawei dongle and any device other than good mobile phones.

With this condition what every I try with OMR may fail.
I need to still research for better cellular sim and restart this experiment.

>>If speedify give you a better result you should use it.
I would like to learn OMR technology and make my own VPN and enjoy my work.
Not interested in 3rd party services.

Ysurac, 2023-02-10:
Do you use a powered USB hub ?

gaamaaresosa, 2023-02-10:
> Do you use a powered USB hub ?

Yes I use Tp-Link powered 8 Port USB hub
I even tried 3 antenna LapCare SIM to LAN adopter.
Even direct Mobile tethering.
All are same disconnection while we do speed test.
I have 5 SIM cards from 3 major cellular providers.

hairy-dog, 2023-02-16:
> Use SQM with download and upload speed set.

Can you clarify what is best for this? I have one vDSL connection plus two or three LTE connections.

For the vDSL I enabled SQM and set the speeds just below the actual speed it manage.

For the LTE connections, I have SQM deselected but selected SQM autorate. I

s this correct, or should SQM be on as well?

Ysurac, 2023-02-16:
SQM autorate only work when SQM is enabled.

hairy-dog, 2023-02-16:
Thank you for the information. If SQM autorate only works with SQM on, should enabling it in the settings also enable SQM?

Ysurac, 2023-02-17:
I need to fix the wizard interface :sweat_smile: