Ysurac/openmptcprouter #2773: Traffic other than TCP, UDP , ICMP?

mtflynn2011, 2023-02-10:
## Expected Behavior
Load balanced IPSEC and/or GRE IP datagrams over WAN links.

## Current Behavior
Currently running OMR with 2 WAN interfaces. MPTCP is filtered on both, so Wireguard is enabled. Shadowsocks is handling TCP. Glorytun-TCP is handling UDP and ICMP. Status page gives "full-green board". IPSEC IP datagrams only use 1 WAN path. TCP is properly load balanced.

Am I missing something? is there a setting to send specific IP datagram types over GLorytun-tcp? 

## Specifications

  - OpenMPTCProuter version: 0.59.1-5.4
  - OpenMPTCProuter VPS version: 0.1028 5.4.207-mptcp
  - OpenMPTCProuter VPS provider:
  - OpenMPTCProuter platform: x86_64<

Ysurac, 2023-02-10:
VPN is used for anything that is not already transported by the proxy (so UDP, ICMP,..).
You can change VPN in the wizard and try what is the best for you or change IPSec configuration to use IPSec over TCP ;)

mtflynn2011, 2023-02-14:
I've rebuilt my setup in case i missed something while debugging and still see the same behavior. See the topology below. 
Notes:

- The firewall limits the VPS side from accessing the general Internet. It allows enough for OMR to connect to the VPS.
- TCP and UDP going from OMR LAN Client to VPS LAN Client load balance across the WAN Interfaces as expected. 
- The VPS and VPS LAN Client have Public IPs.
- IPSEC IP datagrams going from Internal LAN A to Internal LAN B only use one WAN interface on OMR

Any Ideas?

Thanks

[image]

mtflynn2011, 2023-02-14:
I've rebuilt my OMR in case I was missing something from trial-and-error debuging :)

Still seeing the same behavior. 
How does OMR select between sending TCP through ShadowSocks and everything else over whatever VPN is being used. Are there hard-coded OMR-Bypass rules not visible in luci?

My basic topology is below:

Notes:

- OMR WAN interfaces are DHCP from WAN Providers
- VPS and VPS LAN Client have routable Internet IPv4 addresses
- Firewall isolates the VPS LAN from the Internet. It's open enough to allow OMR and the VPS to communicate
- Firewall is what is filtering MPTCP
- TCP and UDP transfers from OMR LAN Client to VPS LAN Client are load balanced by OMR as expected
- IPSEC IP datagrams carrying traffic from Internal LAN A to Internal LAN B only use one WAN interface on OMR

Anyone have any other ideas?

Thanks!

[image]

Ysurac, 2023-02-14:
TCP use the proxy via iptable rule, VPN is the default gateway so used for everything else.

closed 2023-05-21 by github-actions[bot]