Ysurac/openmptcprouter #2776: Setup a VPN interface for OMR-Bypass

iambenmitchell, 2023-02-12:
Hello, I was wondering if anyone has successfully added VPN interfaces? 

I essentially want to add a Windscribe US open vpn connection to OMR and then have OMR send Netflix traffic through it so that I can access US Netflix from the UK. 

However, I tried to figure this out myself and every time I do I end up breaking OMR. I think what that the whole system uses the VPN instead of just having it as an interface. 

Not sure how to do this, might end up just using VLANs and a separate router

iambenmitchell, 2023-02-12:
Basically I want something like this 

**Normal Data:**

Laptop -> OMR Router -> VPS

**Netflix/Other Streaming: **

Laptop -> OMR Bypass -> OpenVPN

Ysurac, 2023-02-15:
You need to make sure interface used for your VPN is not OMRVPN interface (tun0 by default) and that "Use default gateway" is not checked on this interface configuration in Network->Interface.

closed 2023-05-22 by github-actions[bot]