Ysurac/openmptcprouter #2778: Use OMR-Bypass (netflix) with OPNsense

freshwebs, 2023-02-15:
Hello,

i use OPNsense as router and OMR as WAN. 

OMR-Bypass wont work and netflix show vpn-error. 

How can i use OMR as WAN for my OPNsense?

Ysurac, 2023-02-15:
You must use OpenMPTCProuter as DNS to bypass domain with OMR-Bypass, so make a DNS relay in your OPNsense

freshwebs, 2023-02-16:
Thank you, but i does not work. Can you write a simple tutorial?

Ysurac, 2023-02-16:
I don't use OPNsence, so I can't help on this side.

Network-Traditions, 2023-03-06:
I use pfSense, not identical to OPNsense but you should be able to find similar configuration settings.  pfSense is setup as strictly a router with no outbound NAT rules.  With our systems we have pfSense handle all services to all servers, desktops, devices, etc.  DNS is provided by unbound ("DNS Resolver"), which is configured with DNS Query Forwarding enabled.  Then in the general setup, the IP address of the OpenMPTCProuter (OMR) is the only DNS server entered.  This forces pfSense to forward all DNS queries that are not cached or within its local DHCP subnets to OMR.  We primarily use the DOMAINS bypass.  There have been a few times using some of the other bypass options that didn't seem to work properly, but we are unsure of those  conclusions because we used alternate approaches to address achieve our goals of the time.  Hope this helps!

closed 2023-06-10 by github-actions[bot]