Ysurac/openmptcprouter #2781: v0.59.1-5.15-r0+20029-3c06a344e9-x86-64

fareign, 2023-02-17:
use template...
I don't know arch, there is no 0.59.1 on 6.1 kernel,...

_Originally posted by @Ysurac in https://github.com/Ysurac/openmptcprouter/issues/2780#issuecomment-1433593181_
            
hehe,you don't know arch 0.59.1?
where do close the issue quickly?
Or you don't want to solve the problem？

fareign, 2023-02-17:
[image]
[image]
[image]

fareign, 2023-02-17:
[image]
 I solved it.

Ysurac, 2023-02-17:
Please... use template when you open an issue...
I don't want to miss lot of time asking always same questions like version, on what architecture,...
If you solved it, this is close.

closed 2023-02-17 by Ysurac