Ysurac/openmptcprouter #2786: Multipath seems to be blocked on the connection. No Server ping response after 1 second

bagou4502, 2023-02-26:
## Expected Behavior

Hello, 
I got this on my 5g router.
 Multipath seems to be blocked on the connection. No Server ping response after 1 second.
This worked some days ago so i don't understand why this happend only now.
The router is available and work fine.
[image]

## Current Behavior

Yes here are the lasted logs.
```
Sun Feb 26 10:30:08 2023 daemon.notice netifd: Interface 'omrvpn' is now up
Sun Feb 26 10:30:09 2023 daemon.warn [5869]: <warn>  [SQM_IFB_92762] invalid sysfs path read for net/SQM_IFB_92762
Sun Feb 26 10:30:09 2023 daemon.warn [5869]: <warn>  [base-manager] couldn't handle kernel event: device net/SQM_IFB_92762 not found
Sun Feb 26 10:30:09 2023 daemon.warn [5869]: <warn>  [SQM_IFB_b91d3] invalid sysfs path read for net/SQM_IFB_b91d3
Sun Feb 26 10:30:09 2023 daemon.warn [5869]: <warn>  [base-manager] couldn't handle kernel event: device net/SQM_IFB_b91d3 not found
Sun Feb 26 10:30:09 2023 user.notice firewall: Reloading firewall due to ifup of omrvpn (tun0)
Sun Feb 26 10:30:09 2023 user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Sun Feb 26 10:30:09 2023 daemon.err glorytun[19976]: 146.19.168.103.65001: connected
Sun Feb 26 10:30:09 2023 daemon.info glorytun[19976]: STARTED tun0
Sun Feb 26 10:30:09 2023 user.notice mptcp: Reloading mptcp config due to ifup of omrvpn (tun0)
Sun Feb 26 10:30:26 2023 user.notice OMR-VPS: Can't get HarmonyHosting1 token, try later (can ping server HarmonyHosting1 on 37.44.238.152, no server API answer on 37.44.238.152)
```
## Specifications

  - OpenMPTCProuter version v0.59.1-5.4
  - OpenMPTCProuter VPS version: 0.1028 5.4.207-mptcp
  - OpenMPTCProuter VPS provider: HostMyServers,Harmony-Hosting,Sapinet
  - OpenMPTCProuter platform: RPI4 8GB

https://hastebin.skyra.pw/ikabopedun.yaml

Ysurac, 2023-02-26:
The "No Server ping response after 1 second", mean that a "ping -B -I wan1 <yourvpsip>" doesn't work. Try this command via SSH to check result and try another ip, like 1.1.1.1 to check if this connection is usable or not.

bagou4502, 2023-03-01:
Okey i do that asap

bagou4502, 2023-03-01:
Oops missclick

bagou4502, 2023-03-05:
The connexion is usable (I get ping answer) but i see now that the "No server ping" error was removed now i have only "Multipath seems to be blocked on the connection"

closed 2023-03-05 by bagou4502