Ysurac/openmptcprouter #2787: daemon.err glorytun[10647]: couldn't set socket option `TCP_NODELAY'

fareign, 2023-02-27:
## Expected Behavior

## Current Behavior

## Possible Solution

## Steps to Reproduce the Problem

  1.
  2.
  3.

## Context (Environment)

Mon Feb 27 14:39:58 2023 daemon.err glorytun[10647]: 112.74.50.106.65001: connected
Mon Feb 27 14:39:58 2023 daemon.err glorytun[10647]: couldn't set socket option `TCP_NODELAY'
Mon Feb 27 14:39:58 2023 daemon.err glorytun[10647]: couldn't set socket option `TCP_KEEPCNT'
Mon Feb 27 14:39:58 2023 daemon.err glorytun[10647]: couldn't set socket option `TCP_KEEPIDLE'
Mon Feb 27 14:39:58 2023 daemon.err glorytun[10647]: couldn't set socket option `TCP_KEEPINTVL'
Mon Feb 27 14:39:58 2023 daemon.err glorytun[10647]: couldn't set socket option `TCP_USER_TIMEOUT'

## Specifications

  - OpenMPTCProuter version: v0.59.2alpha
  - OpenMPTCProuter VPS version: 0.1028
  - OpenMPTCProuter VPS provider:aliyun
  - OpenMPTCProuter platform:x86_64
  - Country: cn

Ysurac, 2023-02-27:
Not a real problem, and this is an unsupported release for now.

fareign, 2023-02-28:
ok,waiting for 5.15 kernel repair bugs.

closed 2023-06-03 by github-actions[bot]