Ysurac/openmptcprouter #2788: vps doesn't install on amazon aws ec2

rdmitry0911, 2023-03-01:
## Expected Behavior

After installing and final rebooting vps is up and running
## Current Behavior

After installing vps on basic t2.micro instance in amazon aws ec2 with ubuntu server 20.04 and rebooting I can't connect to the instance neither via ssh nor via session manager. Looks like system hangs on booting
## Specifications

  - OpenMPTCProuter version: n/a
  - OpenMPTCProuter VPS version: OpenMPTCProuter Server 0.1028
  - OpenMPTCProuter VPS provider: amazon aws ec2
  - OpenMPTCProuter platform:  x86_64

Ysurac, 2023-03-01:
Make sure you opened ports range on AWS firewall. SSH should then be accessible via port 65222.

rdmitry0911, 2023-03-02:
The port range 0-65535 is opened for both tcp and udp. icmp all protocols is allowed as well according to recommendations for aws from [here](https://github.com/Ysurac/openmptcprouter/wiki/Install-or-update-the-VPS)

rdmitry0911, 2023-03-04:
> ## Expected Behavior
>
> After installing and final rebooting vps is up and running
> ## Current Behavior
>
> After installing vps on basic t2.micro instance in amazon aws ec2 with ubuntu server 20.04 and [...]

I've just checked and it tuned out that UPSTREAM version of vps works in Amazon aws. I would add a note in the vps installation guide about this.

closed 2023-03-04 by rdmitry0911

Ysurac, 2023-03-04:
Upstream release only work with OpenMPTCProuter 5.15/6.1 based release that are in test.

rdmitry0911, 2023-03-05:
> Upstream release only work with OpenMPTCProuter 5.15/6.1 based release that are in test.

What is the difference between upstream and regular release, that prevents the system from booting up in amazon aws?

rdmitry0911, 2023-03-06:
> Upstream release only work with OpenMPTCProuter 5.15/6.1 based release that are in test.

BTW, debian 11 regular vps install script works fine in amazon aws Only ubuntu 20.04 has problems so far.