Ysurac/openmptcprouter #2790: Problème stabilité réseau

JulesCatherine, 2023-03-05:
## Comportement actuel
Salut je vous explique mon problème,
Mon Openmptcprouteur fonctionnait super bien avec le réseau Milkywan rien à dire jusqu'il y a 2 semaines une déconnexion à du avoir lieu je leurs ai donc demandé de me redémarrer ma machine suite à ça j'ai pu me reconnecter tout est bien vert mais le réseau est très instable :
- Appareils connecté domotique ne fonctionne plus
- Connexion Vocal Discord impossible
- Jeux multijoueurs connexion impossible
- Navigation internet très haché

Je me suis donc connecté sur une VM d'une connaissance pour dépanner et tout fonctionne bien réseau super stable et débit bien agrégé.
J'ai fait de même sur une VM Azure le résultat est super bon
(seul problème l'indication pas de réponse au ping sur mes routeurs 4G ainsi que ma Freebox mais pas forcément dérangeant pour le fonctionnement correct.)

Dans le doute j'ai effectué une réinstallation total du système et même constat

Je pense donc que cela vient bien du coté de MilkyWan mais ils m'ont demandé de vous contacter.

Merci pour votre aide.
et Merci Ysurac pour le travail effectué.

  ## Spécifications  
- Version OpenMPTCProuter : 0.59.1-5.4 
- Version VPS OpenMPTCProuter : 0.1028 5.4.207 
- Fournisseur de VPS OpenMPTCProuter : Milkywan, Azure, Orange 
- Plate-forme OpenMPTCProuter : x86_64

Ysurac, 2023-03-06:
Il faut voir dans Status->System log si des erreurs sont détectées.
Sinon il faut vérifier si les DNS d'OpenMPTCprouter sont bien utilisés.

JulesCatherine, 2023-06-07:
Salut, désolé j'avais pas répondu car Milkywan avait réinstaller ma VM et cela avait bien fonctionné mais la ça recommence exactement pareil donc je me permet de relancer le sujet.
Voici mon journal système quand je me connecte sur Milkywan :

-------------------------------------------------------------------------------------------------
Wed Jun  7 19:46:17 2023 user.notice Shadowsocks: Rules DOWN
Wed Jun  7 19:46:18 2023 user.notice Shadowsocks: Rules UP
Wed Jun  7 19:46:18 2023 user.notice Shadowsocks: Reload omr-bypass rules
Wed Jun  7 19:46:18 2023 user.notice omr-bypass: Starting OMR-ByPass...
Wed Jun  7 19:46:19 2023 user.notice OMR-VPS: Get config from server Milkywan...
Wed Jun  7 19:46:19 2023 user.notice omr-bypass: OMR-ByPass is running
Wed Jun  7 19:46:19 2023 user.notice OMR-VPS: Shadowsocks restart...
Wed Jun  7 19:46:19 2023 user.notice Shadowsocks: Rules DOWN
Wed Jun  7 19:46:19 2023 daemon.info /usr/bin/ss-local[1241]: using tcp fast open
Wed Jun  7 19:46:19 2023 daemon.info /usr/bin/ss-local[1241]: enable TCP no-delay
Wed Jun  7 19:46:19 2023 daemon.info /usr/bin/ss-local[1241]: initializing ciphers... chacha20-ietf-poly1305
Wed Jun  7 19:46:19 2023 daemon.info /usr/bin/ss-local[1241]: listening at 127.0.0.1:1111
Wed Jun  7 19:46:19 2023 daemon.info /usr/bin/ss-local[1241]: tcp port reuse enabled
Wed Jun  7 19:46:19 2023 daemon.info /usr/bin/ss-local[1241]: udprelay enabled
Wed Jun  7 19:46:19 2023 daemon.info /usr/bin/ss-local[1241]: udp port reuse enabled
Wed Jun  7 19:46:19 2023 daemon.info /usr/bin/ss-local[1241]: running from root user
Wed Jun  7 19:46:19 2023 user.info : using tcp fast open
Wed Jun  7 19:46:19 2023 daemon.info /usr/bin/ss-redir[1242]: enable TCP no-delay
Wed Jun  7 19:46:19 2023 daemon.info /usr/bin/ss-redir[1242]: resolving hostname to IPv6 address first
Wed Jun  7 19:46:19 2023 daemon.info /usr/bin/ss-redir[1242]: initializing ciphers... chacha20-ietf-poly1305
Wed Jun  7 19:46:19 2023 daemon.info /usr/bin/ss-redir[1242]: listening at :::1100
Wed Jun  7 19:46:19 2023 daemon.info /usr/bin/ss-redir[1242]: tcp port reuse enabled
Wed Jun  7 19:46:19 2023 daemon.info /usr/bin/ss-redir[1242]: UDP relay enabled
Wed Jun  7 19:46:19 2023 daemon.info /usr/bin/ss-redir[1242]: udp port reuse enabled
Wed Jun  7 19:46:19 2023 daemon.info /usr/bin/ss-redir[1242]: running from root user
Wed Jun  7 19:46:19 2023 user.info : using tcp fast open
Wed Jun  7 19:46:19 2023 daemon.info /usr/bin/ss-redir[1243]: enable TCP no-delay
Wed Jun  7 19:46:19 2023 daemon.info /usr/bin/ss-redir[1243]: initializing ciphers... chacha20-ietf-poly1305
Wed Jun  7 19:46:19 2023 daemon.info /usr/bin/ss-redir[1243]: listening at 0.0.0.0:1101
Wed Jun  7 19:46:19 2023 daemon.info /usr/bin/ss-redir[1243]: tcp port reuse enabled
Wed Jun  7 19:46:19 2023 daemon.info /usr/bin/ss-redir[1243]: UDP relay enabled
Wed Jun  7 19:46:19 2023 daemon.info /usr/bin/ss-redir[1243]: udp port reuse enabled
Wed Jun  7 19:46:19 2023 daemon.info /usr/bin/ss-redir[1243]: running from root user
Wed Jun  7 19:46:19 2023 daemon.info glorytun[5972]: STOPPED tun0
Wed Jun  7 19:46:19 2023 daemon.notice netifd: Network device 'tun0' link is down
Wed Jun  7 19:46:19 2023 daemon.notice netifd: Interface 'omrvpn' has link connectivity loss
Wed Jun  7 19:46:19 2023 daemon.notice netifd: Interface 'omrvpn' is now down
Wed Jun  7 19:46:19 2023 daemon.info glorytun: starting glorytun vpn instance vpn
Wed Jun  7 19:46:19 2023 daemon.notice netifd: Interface 'omrvpn' is disabled
Wed Jun  7 19:46:19 2023 daemon.info glorytun[1527]: INITIALIZED tun0
Wed Jun  7 19:46:19 2023 daemon.notice netifd: Interface 'omrvpn' is enabled
Wed Jun  7 19:46:19 2023 daemon.notice netifd: Network device 'tun0' link is up
Wed Jun  7 19:46:19 2023 daemon.notice netifd: Interface 'omrvpn' has link connectivity
Wed Jun  7 19:46:19 2023 daemon.notice netifd: Interface 'omrvpn' is setting up now
Wed Jun  7 19:46:19 2023 daemon.notice netifd: Interface 'omrvpn' is now up
Wed Jun  7 19:46:19 2023 user.notice omr-tracker: Launching...
Wed Jun  7 19:46:20 2023 daemon.err glorytun[1527]: 45.13.XX.XX.65001: connected
Wed Jun  7 19:46:20 2023 daemon.err glorytun[1527]: 45.13.XX.XX.65001: key exchange failed
Wed Jun  7 19:46:20 2023 daemon.info glorytun[1527]: STOPPED tun0
Wed Jun  7 19:46:20 2023 user.notice Shadowsocks: Rules UP
Wed Jun  7 19:46:20 2023 user.notice Shadowsocks: Reload omr-bypass rules
Wed Jun  7 19:46:20 2023 user.notice omr-bypass: Starting OMR-ByPass...
Wed Jun  7 19:46:20 2023 daemon.err glorytun[1527]: 45.13.XX.XX.65001: connected
Wed Jun  7 19:46:21 2023 daemon.err modprobe: failed to find a module named sch_fq_codel
Wed Jun  7 19:46:21 2023 daemon.err glorytun[1527]: 45.13.XX.XX.65001: key exchange failed
Wed Jun  7 19:46:21 2023 daemon.info glorytun[1527]: STOPPED tun0
Wed Jun  7 19:46:21 2023 daemon.warn [8861]: <warn>  [SQM_IFB_8d4f6] invalid sysfs path read for net/SQM_IFB_8d4f6
Wed Jun  7 19:46:21 2023 daemon.warn [8861]: <warn>  [base-manager] couldn't handle kernel event: device net/SQM_IFB_8d4f6 not found
Wed Jun  7 19:46:21 2023 daemon.err modprobe: failed to find a module named sch_fq_codel
Wed Jun  7 19:46:21 2023 daemon.warn [8861]: <warn>  [SQM_IFB_84f5d] invalid sysfs path read for net/SQM_IFB_84f5d
Wed Jun  7 19:46:21 2023 daemon.warn [8861]: <warn>  [base-manager] couldn't handle kernel event: device net/SQM_IFB_84f5d not found
Wed Jun  7 19:46:21 2023 daemon.err modprobe: failed to find a module named sch_fq_codel
[... cut ...]
Wed Jun  7 19:48:21 2023 user.notice post-tracking-post-tracking: wan3 (wan3) switched up
Wed Jun  7 19:48:21 2023 user.notice post-tracking-post-tracking: Interface route not yet set, set route ip r add default via 192.168.2.1 dev wan3 metric 7

-------------------------------------------------------------------------------------------

Merci de votre futur aide car je vois pas trop ce qui coince surtout que la en refaisant une VM chez Sapinet ça refonctionne direct sans faire de modif sur mon routeur.

closed 2023-09-12 by github-actions[bot]