Ysurac/openmptcprouter #2794: Very low speed via vps

rdmitry0911, 2023-03-08:
## Expected Behavior

When running omr-test-speed, expected combined speed should be about 20% less then the combined speed of the individual wan lines
## Current Behavior

omr-test-speed shows the speed of only about 6% of the combined speed of the individual wan lines. At the same time omr-test-speed running on vps shows a speed of about 1Gbps
## Specifications

  - OpenMPTCProuter version:  openmptcprouter v0.59.1-5.4 r0+16594-ce92de8c8c
  - OpenMPTCProuter VPS version:  5.4.207-mptcp 0.1028
  - OpenMPTCProuter VPS provider: Amazon
  - OpenMPTCProuter platform:  RPI4

**Combined speed**
root@OMR-RPI:~# omr-test-speed
Select best test server...
host: scaleway.testdebit.info - ping: 89
host: bordeaux.testdebit.info - ping: 149
host: aix-marseille.testdebit.info - ping: 107
host: lyon.testdebit.info - ping: 99
host: lille.testdebit.info - ping: 91
host: paris.testdebit.info - ping: 88
host: appliwave.testdebit.info - ping: 89
host: speedtest.frankfurt.linode.com - ping: 93
host: speedtest.tokyo2.linode.com - ping: 336
host: speedtest.singapore.linode.com - ping: 245
host: speedtest.newark.linode.com - ping: 163
host: speedtest.atlanta.linode.com - ping: 200
host: speedtest.dallas.linode.com - ping: 199
host: speedtest.fremont.linode.com - ping: 242
host: speed.hetzner.de - ping: 82
host: ipv4.bouygues.testdebit.info - ping: 95
host: par.download.datapacket.com - ping: 104
host: nyc.download.datapacket.com - ping: 173
host: ams.download.datapacket.com - ping: 104
host: fra.download.datapacket.com - ping: 83
host: lon.download.datapacket.com - ping: 109
host: mad.download.datapacket.com - ping: 111
host: prg.download.datapacket.com - ping: 115
host: sto.download.datapacket.com - ping: 121
host: vie.download.datapacket.com - ping: 103
host: war.download.datapacket.com - ping: 109
host: atl.download.datapacket.com - ping: 189
host: chi.download.datapacket.com - ping: 206
host: lax.download.datapacket.com - ping: 238
host: mia.download.datapacket.com - ping: 196
host: nyc.download.datapacket.com - ping: 174
host: speedtest.milkywan.fr - ping: 116
Best server is https://speed.hetzner.de/10GB.bin, running test:
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0  9.7G    0 27.3M    0     0   814k      0  3:29:29  0:00:34  3:28:55  807k^C

**Line 1 speed**
root@OMR-RPI:~# omr-test-speed eth1
Select best test server...
host: scaleway.testdebit.info - ping: 64
host: bordeaux.testdebit.info - ping: 76
host: aix-marseille.testdebit.info - ping: 69
host: lyon.testdebit.info - ping: 70
host: lille.testdebit.info - ping: 76
host: paris.testdebit.info - ping: 57
host: appliwave.testdebit.info - ping: 69
host: speedtest.frankfurt.linode.com - ping: 50
host: speedtest.tokyo2.linode.com - ping: 297
host: speedtest.singapore.linode.com - ping: 290
host: speedtest.newark.linode.com - ping: 152
host: speedtest.atlanta.linode.com - ping: 160
host: speedtest.dallas.linode.com - ping: 180
host: speedtest.fremont.linode.com - ping: 234
host: speed.hetzner.de - ping: 59
host: ipv4.bouygues.testdebit.info - ping: 69
host: par.download.datapacket.com - ping: 71
host: nyc.download.datapacket.com - ping: 144
host: ams.download.datapacket.com - ping: 56
host: fra.download.datapacket.com - ping: 55
host: lon.download.datapacket.com - ping: 66
host: mad.download.datapacket.com - ping: 77
host: prg.download.datapacket.com - ping: 59
host: sto.download.datapacket.com - ping: 71
host: vie.download.datapacket.com - ping: 66
host: war.download.datapacket.com - ping: 78
host: atl.download.datapacket.com - ping: 175
host: chi.download.datapacket.com - ping: 173
host: lax.download.datapacket.com - ping: 221
host: mia.download.datapacket.com - ping: 161
host: nyc.download.datapacket.com - ping: 152
host: speedtest.milkywan.fr - ping: 64
Best server is http://speedtest.frankfurt.linode.com/garbage.php?ckSize=10000, running test:
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 1024M    0 1024M    0     0  11.2M      0 --:--:--  0:01:30 --:--:-- 19.2M

**Line 2 speed**
root@OMR-RPI:~# omr-test-speed eth2
Select best test server...
host: scaleway.testdebit.info - ping: 69
host: bordeaux.testdebit.info - ping: 78
host: aix-marseille.testdebit.info - ping: 82
host: lyon.testdebit.info - ping: 73
host: lille.testdebit.info - ping: 65
host: paris.testdebit.info - ping: 65
host: appliwave.testdebit.info - ping: 68
host: speedtest.frankfurt.linode.com - ping: 70
host: speedtest.tokyo2.linode.com - ping: 314
host: speedtest.singapore.linode.com - ping: 215
host: speedtest.newark.linode.com - ping: 132
host: speedtest.atlanta.linode.com - ping: 151
host: speedtest.dallas.linode.com - ping: 175
host: speedtest.fremont.linode.com - ping: 212
host: speed.hetzner.de - ping: 77
host: ipv4.bouygues.testdebit.info - ping: 76
host: par.download.datapacket.com - ping: 83
host: nyc.download.datapacket.com - ping: 156
host: ams.download.datapacket.com - ping: 87
host: fra.download.datapacket.com - ping: 69
host: lon.download.datapacket.com - ping: 93
host: mad.download.datapacket.com - ping: 91
host: prg.download.datapacket.com - ping: 85
host: sto.download.datapacket.com - ping: 104
host: vie.download.datapacket.com - ping: 60
host: war.download.datapacket.com - ping: 100
host: atl.download.datapacket.com - ping: 162
host: chi.download.datapacket.com - ping: 168
host: lax.download.datapacket.com - ping: 209
host: mia.download.datapacket.com - ping: 168
host: nyc.download.datapacket.com - ping: 145
host: speedtest.milkywan.fr - ping: 79
Best server is http://vie.download.datapacket.com/10000mb.bin, running test:
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
 14 9536M   14 1411M    0     0  5331k      0  0:30:31  0:04:31  0:26:00 5903k^C

**vps speed**
root@debian-vps:/home/admin# omr-test-speed 
Select best test server...
host: speedtest.frankfurt.linode.com - ping: 1
host: speedtest.tokyo2.linode.com - ping: 248
host: speedtest.singapore.linode.com - ping: 162
host: speedtest.newark.linode.com - ping: 82
host: speedtest.atlanta.linode.com - ping: 105
host: speedtest.dallas.linode.com - ping: 119
host: speedtest.fremont.linode.com - ping: 151
host: speedtest.tele2.net - ping: 19
host: speed.hetzner.de - ping: 11
host: ipv4.bouygues.testdebit.info - ping: 10
host: par.download.datapacket.com - ping: 17
host: nyc.download.datapacket.com - ping: 91
host: ams.download.datapacket.com - ping: 7
host: fra.download.datapacket.com - ping: 1
host: lon.download.datapacket.com - ping: 15
host: mad.download.datapacket.com - ping: 33
host: prg.download.datapacket.com - ping: 14
host: sto.download.datapacket.com - ping: 31
host: vie.download.datapacket.com - ping: 18
host: war.download.datapacket.com - ping: 20
host: atl.download.datapacket.com - ping: 107
host: chi.download.datapacket.com - ping: 100
host: lax.download.datapacket.com - ping: 149
host: mia.download.datapacket.com - ping: 118
host: nyc.download.datapacket.com - ping: 87
Best server is http://speedtest.frankfurt.linode.com/garbage.php?ckSize=10000, running test:
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 1024M    0 1024M    0     0  51.2M      0 --:--:--  0:00:19 --:--:-- 76.5M

**OMR iand vps are configured with the default values**
[image]

rdmitry0911, 2023-03-08:
Here it is
[image]

rdmitry0911, 2023-03-08:
> I would need a screenshot of System->OpenMPTCProuter status page. You may need to activate SQM and SQM autorate with download/upload speed set for starlink and cosmoto if it's a mobile connection.

Cosmote is mobile with speed variations 25-70 Mbps , Starlink is sattelite (should be considered as mobile also?) with speed variations 80-350 Mbps How to configure SQM in this case?

Ysurac, 2023-03-08:
You need to configure the best download/upload speeds. SQM autorate will then adapt to real speed.

rdmitry0911, 2023-03-15:
> You need to configure the best download/upload speeds. SQM autorate will then adapt to real speed.

I changed the config to set download/upload  SQM autorotate for both interfaces as recommended in previous post and now I'm monitoring the speed. I noticed that sometimes I have expected combined speed and sometimes poor combined speed. When I have poor combined speed even omr-test-speed running on a specific interface (mobile or satellite in my case) shows poor results. At the same time speedtest running directly on a mobile or on satellite shows normal speed. What else can be tuned?

Ysurac, 2023-03-15:
If you have poor result when using omr-test-speed bind to an interface then you may have an issue outside OMR : USB network card, switch, packets loss,...

kevinh-csalabs, 2023-03-30:
This may be throttling by isp. See if you have the same issues tethering via usb to a laptop at those times. You may also be suffering fragmentation have you checked ping and adjusted MTU if needed?

rdmitry0911, 2023-04-01:
> This may be throttling by isp. See if you have the same issues tethering via usb to a laptop at those times. You may also be suffering fragmentation have you checked ping and adjusted MTU if needed?

I put a 10G file to a host where server part of openmptcprouter is running and shared it with python3 -m  http.server Then I checked the speed by downloading this file with the same command that is used in omr-test-speed:
```
curl -4 -o /dev/null --interface eth1 http://$myhost:$myport/empty.bin || echo
curl -4 -o /dev/null --interface eth2 http://$myhost:$myport/empty.bin || echo

```
I've got quite expected results. About the same as I have using 
```
omr-test-speed eth1 
```
and
```
omr-test-speed eth2
```
That means that there is no problems with the internet path to my host. Nothing is blocked or throttling.
On the other hand side, when I use omr-test-speed in mulipath mode (not binded to a specific interface) to check the speed, I get much slower speed then any of the 2 interface-specifiс speeds. That means, something's wrong either in omr config or in a software, that provides multipathing. And as far as I have default config for omr, something's wrong with the omr software. Am I wrong?

rdmitry0911, 2023-04-01:
BTW, may be provider makes throttling a traffic specific to omr? Is there a way to easy change the ports used by omr?

Ysurac, 2023-04-02:
OMR is not enough know to have any specific throttling.
You can try to set TTL on tethering interfaces, you can also try another tcp congestion in Network->MPTCP

rdmitry0911, 2023-04-03:
> OMR is not enough know to have any specific throttling. You can try to set TTL on tethering interfaces, you can also try another tcp congestion in Network->MPTCP

I can't get the point. IMHO there is no need to change anything in configuration of any of my wan interfaces as I can download the file from my host running vps to omr with a good speed using any of my 2 wan interfaces and without any congestion or throttling problems. However, at the same time I have poor speed while getting the same file over mptcp, that means imho that mptcp works wrong way. It does not combine throughput capabilities of my wan channels. Another possible reason might be that providers throttle my traffic over well-known omr ports. That's why I'm asking if there is a way to change ports used by omr both on server and on client to check if by changing the ports I can improve mptcp download speed? Did I miss any other reason for such a case?

rdmitry0911, 2023-04-05:
> Ports are the same it is just where it is going (could be throttling between you and vps or vps and server vs you and server) It could also be ttl or traffic congestion settings as already mentioned by Ysurac. If you aren't willing to test these then this probably isn't a good fit for you! It is an open source project and works well for many people out of the box but there is no guarantee. I had to do a lot of testing and configuration before I got it working well and stable and still have some issues with ipv6 but it is better than without it!

Look, I do not want to offend anyone here and I know very well how the opensource community works. I really want to understand why some of my tests pass and others don't. Both you and Ysurac suggest to try to change congestion control and ttl. I found some good reviews on this topic and tested the system with both coupled and uncoupled congestion control algorithms. In my case bbr2 is the winner, while changing ttl does not produce a visible effects to the result. I also tried different fullmesh subflows numbers. The default 1 seems to be optimal in my case. With bbr2 I got in avarage about 60% of the combined average speeds of my wan channels. It is much better than earlier, however still less, than expected 80%. Is there anything else I can change to improve mptcp performance?

kevinh-csalabs, 2023-04-05:
If bandwidth is the concern then you are probably observing traffic control. Buffer bloat gets very high on cellular when the connection gets saturated. Is optimize for latency off? Are you using sqm?

rdmitry0911, 2023-04-05:
I tried optimize for latency off and use sqm on. Both options reduce mptcp speed significantly. Up to now the best speed test resalts are when optimize for latency is on and sqm is off

kevinh-csalabs, 2023-04-05:
sqm being on would be expected to reduce speed slightly as it is congestion control and leaves some overhead. Turning optimize for latency off however I would not expect to have a negative effect - in fact the opposite should be true if anything.

Is your raspberry pi firmware up to date? Have you tried adjusting the traffic scheduler (I found better performance with ECF)? Is packet steering enabled under global network options?

rdmitry0911, 2023-04-05:
> sqm being on would be expected to reduce speed slightly as it is congestion control and leaves some overhead. Turning optimize for latency off however I would not expect to have a negative effect - [...]

I'm using the latest omr image [openmptcprouter v0.59.1-5.4 r0+16594-ce92de8c8c] My default combination was BLEST + packet steering = off. I checked all 4 combinations of scheduler and packet steering. In my case the worst is BLEST + packet steering = on, slightly better are BLEST + packet steering = off (my default) and ECF + packet steering = off and even more better is ECF + packet steering = on. Thanks for the tips. Even though the average speed is still less than expected, the peak mptcp speed is sometimes about 100% of the combined speed which has never been possible.

kevinh-csalabs, 2023-04-05:
Good to hear. I wouldn’t expect combined to equal the sum of the individuals. There is always some overhead, but it should be close (80-90% typically, possibly less if optimized to reduce jitter and latency) you may see better overall speed and performance with sqm and auto rate enabled though this won’t be reflected when testing an individual connection and can take a little time for it to adjust itself after enabling.

rdmitry0911, 2023-04-05:
> Good to hear. I wouldn’t expect combined to equal the sum of the individuals. There is always some overhead, but it should be close (80-90% typically, possibly less if optimized to reduce jitter and [...]

I didn't manage to get any improvements with sqm so far. In my case it only makes the things worse for any reason. So I'm going to stay with the current config until a new omr release is ready to install and check. Thanks a lot for your time and help.