Ysurac/openmptcprouter #2799: Help understanding SQM, SQM autorotate and bonding stability over LTE devices

QCWBradleyGebhardt, 2023-03-17:
## Expected Behavior
When running multiple speedtest to get an idea of the usability of the bonded network, I test both the bonded and the individual interfaces and compare the results. I would expect them to be within a few Mbps of each other

## Current Behavior
When comparing the results, I find that adding the individual interface values up they are often times up to 10, and in outlying cases, 20Mbps apart.

To try combat this I tried enabling SQM and SQM autorotate on the LTE interfaces and setting the upload and download values to the averages I had calculated from previous tests done with SQM and SQM autorotate. This resulted in the new results being less than 1Mbps.

I have included a screenshot of a table that includes all the tests I have run. They were done using [speedtest.net](https://www.speedtest.net/) and by disconnecting any interfaces I wasn't using during each test as well as a link to a gist including the router setting, with relevant settings replaced with '********' to keep them private

[Gist with settings](https://gist.github.com/QCWBradleyGebhardt/13aafa558bd54fff114771cf441c110d)

All these finding have led me to the idea that the bonding is exponentially more stable when a decent fibre connection is included, and much less stable when only run on LTE connections

## Specifications

  - OpenMPTCProuter version: openmptcprouter 0.59.1-5.4
  - OpenMPTCProuter VPS version: 0.1028
  - OpenMPTCProuter VPS provider: AWS Debian 11
  - OpenMPTCProuter platform: x86

## Table with results

[image]

Ysurac, 2023-03-17:
Seems that SQM autorate need to be more tuned...
For LTE connections, MPTCP over VPN is a dirty slow workaround that should be used only when MPTCP is blocked.

QCWBradleyGebhardt, 2023-03-22:
Sadly it is blocked on at least 2 of our LTE providers so we're gonna have to use it. By tuned I'm assuming you mean the upload and download values?

Ysurac, 2023-03-22:
No, I mean SQM autorate settings, some are available in Network->SQM QoS.
I will update the script and change some default settings.

closed 2023-06-25 by github-actions[bot]