Ysurac/openmptcprouter #2804: Wan 2 address no defined - no answer to ping server

adrirka, 2023-03-22:
## Expected Behavior

OMR should define an address wan for the second wan (4G/LTE)

## Current Behavior

OMR (xx.xx.100.1) is well setup as the only one dhcp server.
VPS connection is OK : when i check the ip i got the right one (IP of my vps)
Wan1 internet box is well connected  (gateway xx.xx.1.254 and a static ip). The route ip looks good : i check the outpout is made by wan1
But Wan2 4G it does not : setup looks like OK for me with a different gateway xx.xx.2.0 and ip within the same subnet. OMR says : "no answser for the ping server after one second" and there is no adress wan. Internet paquets looks blocked somewhere and cannot pass through.... Is there any specifications to do with the firewall or manual setup with route that i would have missed for OMR side or maybe a wrong setup on router side ?

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.59.1-5.4 r0+16594-ce92de8c
  - OpenMPTCProuter VPS version: Debian 11
  - OpenMPTCProuter VPS provider: OVH
  - OpenMPTCProuter platform: RPI3B+ ARMv8 Processor rev 4

Ysurac, 2023-03-23:
I would need a screenshot of System->OpenMPTCProuter status page.
You can try to ping using this interface via SSH on the router with "ping -B -I <realinterfacename> <VPSIP>" replace <realinterfacename> with the name you can get with "ip a".

adrirka, 2023-03-23:
Please find below the screenshot.
Can you give me more details about ping what you expected?
From terminal router when i use ssh -p 65222 user@VPSipV4 here is the outpout : "ssh-ed25519 host key mismatch for ..."
I can get the ssh connection to the server from a putty client.
[image]

Ysurac, 2023-03-24:
I fixed my previous message.
From status page I can see that the interface is detected as down. It's strange that interface gateway is "192.168.2.0"

adrirka, 2023-03-25:
Many thanks Ysurac for you help. It was due to a gateway issue. OMR has well established the connection right now.
Then now I'm gonna try to fix the ssh key mismatch error that i got when i'm attempting to connect to my vps (with 65222 port) from the terminal router. I didn't define any ssh key so ... don't really understand why.

closed 2023-06-28 by github-actions[bot]