Ysurac/openmptcprouter #2805: EdgeRouter X

abenhamo, 2023-03-23:
Dear all,
Do you have an idea about the speed of aggregation using the edgerouter X ? 

ER-X:
 Dual-Core 880 MHz, MIPS1004Kc , 256 MB DDR3 RAM , which is too slow...

EdgeRouter 4/6/12 has :  4-Core 1GHz MIPS64 Processor and 1Gb DDR3 , is ORM compatible wih ER-4/6/12 ( not the X version) ?

Thank you

Ysurac, 2023-03-26:
ER-X is slow, even interface is quite slow. So aggregated speed is slow too.
For EdgeRouter 4/6/12 I don't have any of them and I support & create image only for arch I own.

abenhamo, 2023-03-27:
Thank you for your reply. Finally i'm mounting a soft router to use ORM x86_64 image i hope it will work as expected, here are the specs:
Pentium Gold 8505
16GB DDR4
256Gb nvme via M.2 2280
Intel i226-V (6 x 2.5gbps)
Wifi6e via M.2 2232

closed 2023-07-01 by github-actions[bot]