Ysurac/openmptcprouter #2806: RPT and UPD port troubles with VoIP (and also telegram and whatsApp audiocalls) using Openmptcprouter

KirillMilchakov, 2023-03-29:
Hello!

Can you help me with port and protocol settings?
I use Openmptcprouter and it works great. But unfortunately I can't call using whatsapp, telegram and my VoIP inside OPMPTCP net.
May be somebody was faced with this trouble?

Having a good day!

KirillMilchakov, 2023-03-30:
Techsupport of my VoIP recommended me to check my nets in terms of routing of SIP batches and I guess my trouble is due to proxing traffic and closed ports for my SIP traffic. So I have an idea may be I can dedicate one wan connection only for sip and do not send traffic to opmptcprouter at all. Is it possible?

Ysurac, 2023-03-30:
There is the option "Enable SIP ALG" in System->OpenMPTCProuter, "Advanced settings" tab

KirillMilchakov, 2023-03-31:
Thank you for the answer, but unfortunately, this option is already works.

Network-Traditions, 2023-04-03:
@KirillMilchakov we operate an Asterisk based PBX over OMR successfully utilizing Google Voice and a carrier grade SIP trunk provider with direct audio.  You can review our latest user experience  in #2392 with [comment](https://github.com/Ysurac/openmptcprouter/issues/2392#issuecomment-1494673367).  Our background experience is available with #2583. I am unfamiliar with whatsapp and telegram protocols, but our experience may provide some insight for your VOIP application.

closed 2023-07-07 by github-actions[bot]