Ysurac/openmptcprouter #2809: Found bugs in 59.1mac address duplication and port mapping failure

jingyilive, 2023-04-02:
I found two bugs in X86 32-bit version using 59.1,
1. When adding a usb Huawei wireless network card, the mac address is duplicated.
2. When port mapping is used, the IP of the destination device cannot be selected, and the manual input is invalid.

Ysurac, 2023-04-02:
1. What you mean ? duplicated with what ?
2. Can you make a screenshot ?

jingyilive, 2023-04-03:
[image]
![10021680503450_ pic](https://user-images.githubuserc
[image]
ontent.com/129647630/229429149-5d5690ec-5e54-4080-a21b-03dbc294c9b7.jpg)
Mon Apr  3 06:18:07 2023 daemon.err modprobe: failed to find a module named sch_fq_codel
Mon Apr  3 06:18:07 2023 daemon.warn [12897]: <warn>  [SQM_IFB_dac42] invalid sysfs path read for net/SQM_IFB_dac42
Mon Apr  3 06:18:07 2023 daemon.warn [12897]: <warn>  [base-manager] couldn't handle kernel event: device net/SQM_IFB_dac42 not found
Mon Apr  3 06:18:07 2023 daemon.err modprobe: failed to find a module named sch_fq_codel
Mon Apr  3 06:18:07 2023 daemon.warn [12897]: <warn>  [SQM_IFB_154ef] invalid sysfs path read for net/SQM_IFB_154ef
Mon Apr  3 06:18:07 2023 daemon.warn [12897]: <warn>  [base-manager] couldn't handle kernel event: device net/SQM_IFB_154ef not found
Mon Apr  3 06:18:07 2023 daemon.err modprobe: failed to find a module named sch_fq_codel
Mon Apr  3 06:18:07 2023 daemon.warn [12897]: <warn>  [SQM_IFB_76028] invalid sysfs path read for net/SQM_IFB_76028
Mon Apr  3 06:18:07 2023 daemon.warn [12897]: <warn>  [base-manager] couldn't handle kernel event: device net/SQM_IFB_76028 not found
Mon Apr  3 06:18:08 2023 daemon.err modprobe: failed to find a module named sch_fq_codel
Mon Apr  3 06:18:08 2023 daemon.warn [12897]: <warn>  [SQM_IFB_0eb40] invalid sysfs path read for net/SQM_IFB_0eb40
Mon Apr  3 06:18:08 2023 daemon.warn [12897]: <warn>  [base-manager] couldn't handle kernel event: device net/SQM_IFB_0eb40 not found
Mon Apr  3 06:18:08 2023 daemon.err modprobe: failed to find a module named sch_fq_codel
Mon Apr  3 06:18:08 2023 daemon.warn [12897]: <warn>  [SQM_IFB_b9fe1] invalid sysfs path read for net/SQM_IFB_b9fe1
Mon Apr  3 06:18:08 2023 daemon.warn [12897]: <warn>  [base-manager] couldn't handle kernel event: device net/SQM_IFB_b9fe1 not found
Mon Apr  3 06:18:08 2023 daemon.err modprobe: failed to find a module named sch_fq_codel
Mon Apr  3 06:18:08 2023 daemon.warn [12897]: <warn>  [SQM_IFB_53cc6] invalid sysfs path read for net/SQM_IFB_53cc6
Mon Apr  3 06:18:08 2023 daemon.warn [12897]: <warn>  [base-manager] couldn't handle kernel event: device net/SQM_IFB_53cc6 not fou

jingyilive, 2023-04-03:
[image]

Ysurac, 2023-04-04:
You can set MAC address in Network->Interface, "Devices" tab, then "address mac". If nothing is defined, then the mac come from your USB device. What is the model ?
You should be able to enter any valid IP address as Internal IP address for Ports forwards, what error do you have ?

jingyilive, 2023-04-11:
Manual entry is invalid, mapping was unsuccessful

closed 2023-07-16 by github-actions[bot]