Ysurac/openmptcprouter #2810: Can acces irp.nain-t.net -- OMR problème

vinceducat, 2023-04-02:
## Expected Behavior
acces web site irp.nain-t.net

## Current Behavior
can't acces irp.nain-t.net 
even i change dns to 1.1.1.1 or bypass domain .. in OMR router
change machine too...
check with my isp milkywan, all is good through there network check with support

Need some help to find a solution ;-)

## Specifications

  - OpenMPTCProuter version: 0.59.1
  - OpenMPTCProuter VPS version: 0.1028
  - OpenMPTCProuter VPS provider: Milkywan.fr
  - OpenMPTCProuter platform: x86_64

vinceducat, 2023-04-02:
2a0e:e701:1129:0:899c:d5b:9dd1:a717/51706 query[A] irp.nain-t.net from 2a0e:e701:1129:0:899c:d5b:9dd1:a717
Sun Apr  2 19:46:49 2023 daemon.info dnsmasq[1]: 251 2a0e:e701:1129:0:899c:d5b:9dd1:a717/51706 forwarded irp.nain-t.net to 127.0.0.1
Sun Apr  2 19:46:49 2023 daemon.info dnsmasq[1]: 251 2a0e:e701:1129:0:899c:d5b:9dd1:a717/51706 reply irp.nain-t.net is <CNAME>
Sun Apr  2 19:46:49 2023 daemon.info dnsmasq[1]: 251 2a0e:e701:1129:0:899c:d5b:9dd1:a717/51706 reply vps.nain-t.net is 51.68.121.59
Sun Apr  2 19:46:49 2023 daemon.info dnsmasq[1]: 252 2a0e:e701:1129:0:899c:d5b:9dd1:a717/51935 query[AAAA] irp.nain-t.net from 2a0e:e701:1129:0:899c:d5b:9dd1:a717
Sun Apr  2 19:46:49 2023 daemon.info dnsmasq[1]: 252 2a0e:e701:1129:0:899c:d5b:9dd1:a717/51935 cached irp.nain-t.net is <CNAME>
Sun Apr  2 19:46:49 2023 daemon.info dnsmasq[1]: 252 2a0e:e701:1129:0:899c:d5b:9dd1:a717/51935 forwarded irp.nain-t.net to 127.0.0.1
Sun Apr  2 19:46:49 2023 daemon.info dnsmasq[1]: 252 2a0e:e701:1129:0:899c:d5b:9dd1:a717/51935 reply irp.nain-t.net is <CNAME>
Sun Apr  2 19:46:49 2023 daemon.info dnsmasq[1]: 252 2a0e:e701:1129:0:899c:d5b:9dd1:a717/51935 reply vps.nain-t.net is 2001:41d0:305:2100::2cd5

Ysurac, 2023-04-02:
What is the result of a traceroute ?

vinceducat, 2023-04-02:
C:\Users\vincent>tracert irp.nain-t.net

Détermination de l’itinéraire vers vps.nain-t.net [2001:41d0:305:2100::2cd5]
avec un maximum de 30 sauts :

  1     2 ms     2 ms     2 ms  OpenMPTCProuter.lan [2a0e:e701:1129::1]
  2    47 ms    45 ms    52 ms  OpenMPTCProuter.lan [2a0e:e701:1129::1]
  3    48 ms    43 ms    45 ms  one-gw-to-rule-them-all.cust.milkywan.net [2a0b:cbc0:10:1::1]
  4    45 ms    44 ms    45 ms  te2.a7050sx.edge.th2.bb.ip4.milkywan.net [2a0b:cbc0:1::c1]
  5    47 ms    45 ms    46 ms  fo36.a7050qxs.edge.th2.bb.ip6.milkywan.net [2a0b:cbc0:1::192]
  6    45 ms    47 ms    48 ms  te1.2985.a7050sx.edge.dc2.bb.ip6.milkywan.net [2a0b:cbc0:1::91]
  7   391 ms     *      213 ms  ovh.par.franceix.net [2001:7f8:54::144]
  8     *        *        *     Délai d’attente de la demande dépassé.
  9     *        *        *     Délai d’attente de la demande dépassé.
 10    53 ms    53 ms    54 ms  vl100.gra-d1-a75.fr.eu [2001:41d0::42d]
 11     *        *        *     Délai d’attente de la demande dépassé.
 12    54 ms    56 ms    54 ms  2001:41d0:0:50::1:9971
 13    56 ms    53 ms    53 ms  2001:41d0:0:1:3:0:1:4715
 14    55 ms    53 ms    62 ms  2001:41d0:0:1:3:0:1:49e7
 15    56 ms    53 ms    54 ms  2001:41d0:0:1:3:0:1:4c08
 16     *        *        *     Délai d’attente de la demande dépassé.
 17    53 ms    54 ms    52 ms  smtp.nain-t.net [2001:41d0:305:2100::2cd5]

vinceducat, 2023-04-02:
yes with with https:// and 51.68.121.59 => web browser goes on but with certificat warning 
https://51.68.121.59/doku.php/start 

same problem and work in ipv6 for https://lazarus-ide.org https://37.97.187.115/

Ysurac, 2023-04-02:
Do you have any result using "curl -6 irp.nain-t.net" on the router via SSH and same on the VPS ?

vinceducat, 2023-04-02:
on OMR
root@OpenMPTCProuter:~# curl -6 irp.nain-t.net

command block :-( no return do ctrl+c to quit

vinceducat, 2023-04-02:
OMR VPS
vincentm@vincentm:~$ sudo curl -6 irp.nain-t.net
sudo: unable to resolve host vincentm: Name or service not known

only this line and block too

Ysurac, 2023-04-02:
The problem doesn't seems to be on OMR side.
Try a "curl -vvvv -6 irp.nain-t.net" on VPS (no need to sudo) to check if you have more info.

Ysurac, 2023-04-02:
I checked from another VPS, the website doesn't seems to answer on IPv6.

vinceducat, 2023-04-02:
vincentm@vincentm:~$ curl -6 irp.nain-t.net
block too

vinceducat, 2023-04-02:
ok @Ysurac strange ... test on my smartphone and seems ok ... it the smartphone not uses ipv6 ?

vinceducat, 2023-04-02:
it there a way to solve this ?

Ysurac, 2023-04-02:
You should be able to disable IPv6 for a this domain by adding "/irp.nain-t.net/::" in Network->DHCP&DNS, DNS forward.
(You may have a DNS cache on your client computer that will make time before doing a new request)

vinceducat, 2023-04-02:
add this line under /use-application-dns.net/ ? with the double ":" ??

closed 2023-07-10 by github-actions[bot]