Ysurac/openmptcprouter #2811: Ubuntu 22.04 VPS crash after installing script

finnikola, 2023-04-03:
## Expected Behavior
Install VPS script

## Current Behavior
Hi, I am not very experienced and am having a problem installing the vps script.
I have purchased an Ubuntu 22.04 VPS and when I install the script it gets to the end and makes me reboot.

However at this point the vps fails to reboot and I have to have it totally restored by the provider.
This has already happened twice.

From the console I see the following screenshot.
[image]

- first time I tried the ubuntu 20.04 script (I don't see 22.04)
- second time I tried with debian.

The same problem always occurs.

Where am I going wrong?

## Specifications

  - OpenMPTCProuter version: 0.59.1
  - OpenMPTCProuter VPS version: Ubuntu 22.04
  - OpenMPTCProuter VPS provider: ServerPlan
  - OpenMPTCProuter platform: X86_64

Ysurac, 2023-04-04:
I need to test on a ServerPlan VPS, what the VPS you took ?

finnikola, 2023-04-05:
Hi, i took this VPS:
[image]

I ask serverplan to restart vps with ubuntu 20.04 an now script installed sucessfully.

Melania95, 2023-04-22:
Hi to everyone, i have same problem. I have purchased a Ubuntu 22.04 Vps and not work i have same scree like finnikola user...OpenMPTCProuter work with Ubuntu 22.04 or only with 20.04 version? Someone can help me? Thanks in advance.

finnikola, 2023-04-22:
> Hi to everyone, i have same problem. I have purchased a Ubuntu 22.04 Vps and not work i have same scree like finnikola user...OpenMPTCProuter work with Ubuntu 22.04 or only with 20.04 version? Someone can help me? Thanks in advance.

Hello,
I can't tell you if it works with 22.04 because I then had a new 20.04 vps reinstalled and now everything works perfectly.

Melania95, 2023-04-24:
> Hello, I can't tell you if it works with 22.04 because I then had a new 20.04 vps reinstalled and now everything works perfectly.
Thanks for your reply, now i have changed my Vps with Ubuntu 20.4 (like you Serverplan) i have installed all but don't work...i have tried some Vps Debian 10 and no problem but now i have inserted my Ip and my Server Key and no work. Is there something special to do or set with Ubuntu 20.4 and Vps Serverplan? Can you help me? Thank you

Melania95, 2023-04-25:
> I need to test on a ServerPlan VPS, what the VPS you took ?

Hi Ysurac, i bought a Vps from Serverplan with O.S. Ubuntu 20.4, so I typed: "wget ​​-O - https://www.openmptcprouter.com/server/ubuntu20.04-x86_64.sh | sh" the installation is fine but it doesn't work. I tried several Vps with Debian 10 and no problem but now it doesn't work. Do I have to open ports on the router (I use the Linksys wrt3200acm)? Could you help me? Thanks. Melania.

gstammw, 2023-05-27:
So we need support for Ubuntu 22.04, don't we?

closed 2023-09-01 by github-actions[bot]