Ysurac/openmptcprouter #2812: How to switch to a backup server when primary server fails

iamdivyanshverma, 2023-04-04:
## Expected Behavior

## Current Behavior

## Specifications

  - OpenMPTCProuter version: 
  - OpenMPTCProuter VPS version: 
  - OpenMPTCProuter VPS provider:
  - OpenMPTCProuter platform:

Ysurac, 2023-04-08:
This can be done via "uci" command and replacing VPS IP by another.

iamdivyanshverma, 2023-04-10:
Hi, @Ysurac Thanks for the update..

I tried with 
uci set openmptcprouter.vps.ip='xxx.xxx.xxx.xxx'
uci set openmptcprouter.vps.password='xxxxxxx'
uci commit openmptcprouter

but it does not sync to the new server ..!!
if i am missing additional steps please let me know...!!

kmrinal, 2023-04-30:
@Ysurac Updating the vps ip and vps password does not seem to help. 

We updated the values in openmptcprouter.vps.ip and openmptcprouter.vps.password but the omr is still talking to the old server. 
In fact, even after restart, the omr device is still connected to the old server.

Do we need to make some other changes or perhaps call some internal API to make it update its connections?

Ysurac, 2023-04-30:
```uci show | grep <ip>``` and replace all.

iamdivyanshverma, 2023-06-15:
Hi @Ysurac ,

I tried replacing IP address with a shell script.
uci set dsvpn.vpn.host="$1"
uci set glorytun.vpn.host="$1"
uci set glorytun-udp.vpn.host="$1"
uci set iperf.vps.host="$1"
uci set mlvpn.general.host="$1"
uci set openmptcprouter.vps.port='65500'
uci set openmptcprouter.vps.username="openmptcprouter"
uci set openmptcprouter.vps.ip="$1"
uci set openmptcprouter.omr.detected_ss_ipv4="$1"
uci set openmptcprouter.omr.detected_public_ipv4="$1"
uci set openmptcprouter.vps.password="$2"
uci set openvpn.omr.remote="$1"
uci set shadowsocks-libev.sss0.obfs_host="$1"
uci set shadowsocks-libev.sss0.server="$1"
uci set shadowsocks-libev.sss1.obfs_host="$1"
uci set shadowsocks-libev.sss1.server="$1"
uci set upnpd.config.external_ip="$1"
uci set v2ray.omrout.s_vmess_address="$1"
uci set v2ray.omrout.s_vless_address="$1"
uci commit

I rebooted the system but still I am not able to connect to the new server, in logread it's coming like cant get VPS token.

root@Zifilink:~# logread
Thu Jun 15 07:01:45 2023 kern.info kernel: [    0.273596] RAPL PMU: hw unit of domain dram 2^-14 Joules
Thu Jun 15 07:01:45 2023 kern.info kernel: [    0.273601] RAPL PMU: hw unit of domain pp1-gpu 2^-14 Joules
Thu Jun 15 07:01:45 2023 kern.info kernel: [    0.273666] clocksource: tsc: mask: 0xffffffffffffffff max_cycles: 0xfc66f4fc7c, max_idle_ns: 440795224246 ns
Thu Jun 15 07:01:45 2023 kern.info kernel: [    0.273745] clocksource: Switched to clocksource tsc
Thu Jun 15 07:01:45 2023 kern.info kernel: [    0.275236] workingset: timestamp_bits=46 max_order=20 bucket_order=0
Thu Jun 15 07:01:45 2023 kern.info kernel: [    0.278340] squashfs: version 4.0 (2009/01/31) Phillip Lougher
Thu Jun 15 07:01:45 2023 kern.info kernel: [    0.278786] Block layer SCSI generic (bsg) driver version 0.4 loaded (major 249)
Thu Jun 15 07:01:45 2023 kern.info kernel: [    0.281991] hv_vmbus: registering driver hyperv_fb
Thu Jun 15 07:01:45 2023 kern.debug kernel: [    0.282019] intel_idle: Please enable MWAIT in BIOS SETUP
Thu Jun 15 07:01:45 2023 kern.info kernel: [    0.282296] input: Power Button as /devices/LNXSYSTM:00/LNXSYBUS:00/PNP0C0C:00/input/input0
Thu Jun 15 07:01:45 2023 kern.info kernel: [    0.282417] ACPI: Power Button [PWRB]
Thu Jun 15 07:01:45 2023 kern.info kernel: [    0.282512] input: Power Button as /devices/LNXSYSTM:00/LNXPWRBN:00/input/input1
Thu Jun 15 07:01:45 2023 kern.info kernel: [    0.282602] ACPI: Power Button [PWRF]
Thu Jun 15 07:01:45 2023 kern.info kernel: [    0.288178] thermal LNXTHERM:00: registered as thermal_zone0
Thu Jun 15 07:01:45 2023 kern.info kernel: [    0.288191] ACPI: Thermal Zone [TZ01] (46 C)
Thu Jun 15 07:01:45 2023 kern.info kernel: [    0.288640] Serial: 8250/16550 driver, 16 ports, IRQ sharing enabled
Thu Jun 15 07:01:45 2023 kern.info kernel: [    0.291033] Non-volatile memory driver v1.3
Thu Jun 15 07:01:45 2023 kern.info kernel: [    0.291234] Linux agpgart interface v0.103
Thu Jun 15 07:01:45 2023 kern.info kernel: [    0.292400] [drm] Supports vblank timestamp caching Rev 2 (21.10.2013).
Thu Jun 15 07:01:45 2023 kern.info kernel: [    0.292410] [drm] Driver supports precise vblank timestamp query.
Thu Jun 15 07:01:45 2023 kern.warn kernel: [    0.293889] i915 0000:00:02.0: Direct firmware load for i915/bxt_dmc_ver1_07.bin failed with error -2
Thu Jun 15 07:01:45 2023 kern.warn kernel: [    0.293902] i915 0000:00:02.0: Falling back to sysfs fallback for: i915/bxt_dmc_ver1_07.bin
Thu Jun 15 07:01:45 2023 kern.info kernel: [    0.297748] [drm] Initialized i915 1.6.0 20190822 for 0000:00:02.0 on minor 0
Thu Jun 15 07:01:45 2023 kern.info kernel: [    0.299956] ACPI: Video Device [GFX0] (multi-head: yes  rom: no  post: no)
Thu Jun 15 07:01:45 2023 kern.info kernel: [    0.300572] input: Video Bus as /devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/LNXVIDEO:00/input/input2
Thu Jun 15 07:01:45 2023 kern.info kernel: [    0.303511] [drm] Cannot find any crtc or sizes
Thu Jun 15 07:01:45 2023 kern.info kernel: [    0.303840] loop: module loaded
Thu Jun 15 07:01:45 2023 kern.info kernel: [    0.303915] Guest personality initialized and is inactive
Thu Jun 15 07:01:45 2023 kern.info kernel: [    0.303998] VMCI host device registered (name=vmci, major=10, minor=63)
Thu Jun 15 07:01:45 2023 kern.info kernel: [    0.304005] Initialized host personality
Thu Jun 15 07:01:45 2023 kern.notice kernel: [    0.304079] lpc_ich 0000:00:1f.0: I/O space for ACPI uninitialized
Thu Jun 15 07:01:45 2023 kern.info kernel: [    0.304223] Loading iSCSI transport class v2.0-870.
Thu Jun 15 07:01:45 2023 kern.info kernel: [    0.304424] VMware PVSCSI driver - version 1.0.7.0-k
Thu Jun 15 07:01:45 2023 kern.info kernel: [    0.304457] hv_vmbus: registering driver hv_storvsc
Thu Jun 15 07:01:45 2023 kern.debug kernel: [    0.305885] ahci 0000:00:12.0: version 3.0
Thu Jun 15 07:01:45 2023 kern.info kernel: [    0.306348] ahci 0000:00:12.0: AHCI 0001.0301 32 slots 2 ports 6 Gbps 0x3 impl SATA mode
Thu Jun 15 07:01:45 2023 kern.info kernel: [    0.306365] ahci 0000:00:12.0: flags: 64bit ncq sntf clo only pmp pio slum part deso sadm sds apst
Thu Jun 15 07:01:45 2023 kern.info kernel: [    0.306955] scsi host0: ahci
Thu Jun 15 07:01:45 2023 kern.info kernel: [    0.307342] scsi host1: ahci
Thu Jun 15 07:01:45 2023 kern.info kernel: [    0.307452] ata1: SATA max UDMA/133 abar m2048@0x9141b000 port 0x9141b100 irq 124
[... cut ...]
Thu Jun 15 07:02:56 2023 user.notice OMR-VPS: Can't get vps token, try later ()
Thu Jun 15 07:03:06 2023 user.notice OMR-VPS: Can't get vps token, try later ()
Thu Jun 15 07:03:15 2023 user.notice OMR-VPS: Can't get vps token, try later ()
Thu Jun 15 07:03:25 2023 user.notice OMR-VPS: Can't get vps token, try later ()
Thu Jun 15 07:03:35 2023 user.notice OMR-VPS: Can't get vps token, try later ()
Thu Jun 15 07:03:44 2023 user.notice OMR-VPS: Can't get vps token, try later ()
Thu Jun 15 07:03:54 2023 user.notice OMR-VPS: Can't get vps token, try later ()

closed 2023-09-19 by github-actions[bot]