Ysurac/openmptcprouter #2815: Random slow performance, reboot solves it.

IVCrossway, 2023-04-07:
## Expected Behavior
Boding speed should be around 160 down - 40 up.

## Current Behavior
Download speed is about 30/40 mbps, upload is terrible 1~5mbps.
Rebooting OMR solves the problem.

## Specifications

  - OpenMPTCProuter version: 0.59.1-5.4
  - OpenMPTCProuter VPS version: 0.1028 5.4.207-mptcp
  - OpenMPTCProuter VPS provider: Aruba (Italy)
  - OpenMPTCProuter platform: Lenovo ThinkCenter Tiny M710q

Log file: https://pastebin.com/0hKkTRge
[image]
[image]
[image]
[image]
[image]
[image]

Ysurac, 2023-04-08:
This is the issue:
```
MPTCP: out of memory: force closing socket
Fri Apr  7 17:18:14 2023 kern.info kernel: [165709.321383] TCP: too many orphaned sockets
```
Do you use some p2p or anything that can create many connections ?
How much RAM do you have ?

IVCrossway, 2023-04-08:
> 

The only "intensive" network activity I do is sync files from my pc to Onedrive using Rclone, but it only transfers 4 files at a time. The mini pc hosting OMR is equipped with 4gb of ram

winkmichael, 2023-04-08:
Are you out of memory, can you create a swap drive?

IVCrossway, 2023-04-08:
Yes, I've got plenty of ssd space free to create a swap file, but the problem is that when this happens the ram is not even full... I've never seen OMR using more than 1GB of ram

Ysurac, 2023-04-14:
Via SSH on the router, what is the result of ```sysctl net.ipv4.tcp_mem```, ```sysctl net.ipv4.tcp_max_orphans```, ```sysctl net.ipv4.tcp_keepalive_time```, ```net.ipv4.tcp_keepalive_intvl``` and ```sysctl net.ipv4.tcp_keepalive_probes``` ?

IVCrossway, 2023-04-14:
```
root@omr:~# sysctl net.ipv4.tcp_mem
net.ipv4.tcp_mem = 8092 131072  67108864
```

```
root@omr:~# sysctl net.ipv4.tcp_max_orphans
net.ipv4.tcp_max_orphans = 16384
```

```
root@omr:~# sysctl net.ipv4.tcp_keepalive_time
net.ipv4.tcp_keepalive_time = 7200
```

```
root@omr:~# sysctl net.ipv4.tcp_keepalive_intvl
net.ipv4.tcp_keepalive_intvl = 75
```

```
root@omr:~# sysctl net.ipv4.tcp_keepalive_probes
net.ipv4.tcp_keepalive_probes = 9
```

closed 2023-07-18 by github-actions[bot]