Ysurac/openmptcprouter #2816: Rooted smartphone support?

openaspace, 2023-04-08:
Hi.
It's possible to install openmptcproutet client on Android hardware phones with dual SIM card and USBC to Ethernet adapter??!? 

Any successful installation on some smartphone hardware?

I need to use it for video streaming giving connection to a video camera.

Thanks

Ysurac, 2023-04-08:
OpenMPTCProuter can't be installed on any phone. It's a router and need a hardware that can work as a router.
It can then be used by a phone, a pc or anything via wifi or ethernet like any router.

openaspace, 2023-04-08:
Yes.
I mean to connect to the remote server, using the phone dual SIM as client and external device connected to the USBC in thetering.
Is this possible?

Ysurac, 2023-04-08:
Using the phone as a WAN via USB tethering ? yes it's possible. Should be shown as an USB interface in OpenMPTCProuter and can be used.
If you have issue on this, you can follow OpenWRT documentations.

openaspace, 2023-04-08:
> Using the phone as a WAN via USB tethering ? yes it's possible. Should be shown as an USB interface in OpenMPTCProuter and can be used.
> If you have issue on this, you can follow OpenWRT documentations.

No i mean some solution to have the redundancy connection of the phone using the openwrt local router as gateway to the remote bonding server.

To be clear:
- I have a video camera with Ethernet connection.
- I want to share smartphone dual SIM connection in backup mode with the camera over USBC Ethernet.

Considering we can't install the router directly on the phone, is there a solution to take advantage of the dual SIM connection in the openmtcp environment?

No way to do this?

Ysurac, 2023-04-08:
If the phone can failover on the other SIM himself yes. But both SIMs can be used at same times (would be even better).

openaspace, 2023-04-08:
> If the phone can failover on the other SIM himself yes. But both SIMs can be used at same times (would be even better).

But if the android doesn't support the openwrt approach for the double connection, how I can connect my double SIM phone to the openmtcp router server model?

ccmks, 2023-06-15:
For smartphone, the only option is to install Speedify, with rooted, you can combine Cellular data and Wifi then use USB to tether. However, the result might not great since speedify is public server and speed might not be fast enough

closed 2023-09-19 by github-actions[bot]