Ysurac/openmptcprouter #2818: CORS Policy not allowing Firmware Update

alessandromrc, 2023-04-13:
## Expected Behavior

I expected the download and install of a new firmware update

## Current Behavior

CORS Policy is blocking the API call

## Possible Solution

Probably checking CORS Policy settings again

## Steps to Reproduce the Problem

  1. Go to /cgi-bin/luci/admin/system/sysupgrade
  2. Press on "Request Firmware"
  
[image]

## Context (Environment)

## Specifications

  - OpenMPTCProuter version: v0.59.1
  - OpenMPTCProuter VPS version: v0.1028
  - OpenMPTCProuter VPS provider: Server Plan
  - OpenMPTCProuter platform: RPI4 (8GB)
  - Country: IT

AlexEran, 2023-09-09:
Today I tried to update OpenMPTCProuter from 0.58.5 and got the same problem. How to resolve the problem?

[image]

Ysurac, 2023-09-09:
You can download the image from https://www.openmptcprouter.com/download and update with System->Backup/Flash Firmware, and use "Flash Image..." button.
I will try to find and fix the issue ASAP, maybe a problem with the CDN.

alessandromrc, 2023-09-09:
> You can download the image from https://www.openmptcprouter.com/download and update with System->Backup/Flash Firmware, and use "Flash Image..." button. I will try to find and fix the issue ASAP, maybe a problem with the CDN.

I doubt it is a CDN issue but probably just a bad configuration of CORS that happens 99.9% as it kinda suck ;)