Ysurac/openmptcprouter #2821: OMR Not aggregating connections

gifive, 2023-04-19:
## Expected Behavior
I have 3 internet connections connected to my OMR installation. I expect to aggregate the 3 to have, more or less, a total speed equal to the sum of the 3. 

## Current Behavior
Only one connection is used at a time. If i run a speedtest, the result is equal to the speed of the fastest connection.
In System -> OMR -> Status everything seems fine.
I also tried MPTCP over VPN, nothing changes.
My public IP is the VPS one, so connection to the VPS works. Fallback works.
My VPS is on AWS.

## Specifications

  - OpenMPTCProuter version: 0.59.1-5.4
  - OpenMPTCProuter VPS version: 0.1028 5.15.57-mptcp
  - OpenMPTCProuter VPS provider: AWS
  - OpenMPTCProuter platform: x86

Ysurac, 2023-04-19:
Your VPS is using a 5.15 release and you router a 5.4 release, you should have same kernel on both end. There is know problems on current release when using 5.4 router and 5.15 VPS.

gifive, 2023-04-19:
thank you very much @Ysurac for the reply. i think the issue is that i selected the upstream version during the vps installation. What is the difference between 5.4 and 5.15?

Ysurac, 2023-04-19:
5.4 use out of tree MPTCP and 5.15 use MPTCP in kernel. It's not same implementation of MPTCP.
I'm working on MPTCP using 6.1 kernel (it's working better than 5.15).

Ysurac, 2023-04-25:
For now it's still 5.4.

closed 2023-07-29 by github-actions[bot]

hhaniel, 2024-04-13:
I got the router running on vmware and vps on debian and upgraded to the latest 60-61 release and getting this problem - i have even started configuring everything from scratch and I am getting this problem - also all the MPTCP test come back with me not using MOTCP - I can’t work out where i am going wrong - everything was working finge on version 59

Ysurac, 2024-04-13:
@hhaniel it's an old issue, open a new one.

hhaniel, 2024-04-13:
I have commented on #3225 which is open and similar