Ysurac/openmptcprouter #2822: 4 or 5G USB dongle

djatomix, 2023-04-19:
Good morning

I'm currently looking for a 4G or 5G USB dongle to add to the OMR setup I just ran in x86_64. 
(if possible, on Amazon FR shop)

I saw the EM7565 modules: https://www.amazon.fr/gp/product/B0BQFNQQMW/ref=ox_sc_act_title_1?smid=A2GY03IJEBMI1H&psc=1
and this box: https://www.amazon.fr/gp/product/B09HY1XSWF/ref=ox_sc_act_title_5?smid=A2GNN9CKTV5QUI&psc=1

What do you think ?

Thanks in advance.

(sorry for my English, I go through a translator...)

  - OpenMPTCProuter version: v0.59.1-5.4-r0+16594
  - OpenMPTCProuter VPS provider: OVH
  - OpenMPTCProuter platform: x86_64

Ysurac, 2023-04-19:
I have no idea about these. 
I have an old Huawei E5186s-22a and some Huawei E8372h-155 USB dongle. Both are working correctly in France.

ZonaLTE, 2023-04-20:
I made such a board, and I'm trying various modems...
[image]

dfadev, 2023-04-20:
I use two of these:
https://www.aliexpress.us/item/3256804396044925.html

These probably work too:
https://www.aliexpress.us/item/3256804442228703.html

closed 2023-07-26 by github-actions[bot]