Ysurac/openmptcprouter #2835: Setting up Teltonika RUTX09 for OpenMPTCProuter

andreamichelon, 2023-04-28:
## Expected Behavior
Trying to set up two Teltonika RUTX09 to connect to OpenMPTCProuter. I have setup the Teltonika using a static IP and disabling the DHCP.
[image]

## Current Behavior
On OpenMPTCProuter I get No Server ping response after 1 second. This is how is setup.

[image]
[image]

## Specifications

  - OpenMPTCProuter version: v0.59.1-5.4
  - OpenMPTCProuter VPS version: v0.59.1-5.4
  - OpenMPTCProuter VPS provider: DigitalOcean
  - OpenMPTCProuter platform: x86_64

Thank you for your help

Ysurac, 2023-04-28:
Configuration seems to be ok, make sure "Use default gateway" is disabled in advanced settings tab of interface config.
You should use the wizard to configure interfaces/key/...

andreamichelon, 2023-04-29:
Hi Yusrac, I managed to solve the issue. It was a wrong setting in the Teltonika.

Now, everything seems to work apart from the VPS, where I get:

Can't get public IP address from ShadowSocks

[image]

I tried to look on the forum for a possible solution, but I couldn't find a straight answer. 
VPS is on a Digital Ocean droplet with a Derbian 11 x68
Version is VPS 0.1028

Can you help? Thank you,

Ysurac, 2023-04-29:
The VPS kernel is not the good one, did you reboot the VPS after script installation ?

andreamichelon, 2023-04-29:
Yes, I did. What can I do to give you a better picture of the problem?

Ysurac, 2023-04-29:
What is the result of ```uname -a``` on the VPS ?

Ysurac, 2023-04-29:
I just launched the script on a Digital Ocean droplet with Debian 11 x86_64 and after reboot I get the correct kernel: ```Linux debian-s-1vcpu-512mb-10gb-fra1-01 5.4.207-mptcp```

andreamichelon, 2023-05-01:
Hi Ysurac, sorry for the late reply. Running `uname -a` I get:
`Linux omptcpr-VPS 5.10.0-11-amd64 #1 SMP Debian 5.10.92-2 (2022-02-28) x86_64 GNU/Linux`

Ysurac, 2023-05-02:
Try to run the VPS install script again.

closed 2023-08-06 by github-actions[bot]