Ysurac/openmptcprouter #2837: v2ray-plugin will not be available on next release.

kmrinal, 2023-04-29:
v2ray-plugin will not be available on next release.

_Originally posted by @Ysurac in https://github.com/Ysurac/openmptcprouter/issues/2828#issuecomment-1522260057_

@Ysurac Does that mean UDP traffic will be handled differently now?

Kalimeiro, 2023-04-29:
As has always been the case, UDP goes through the VPN, so through GloryTun.

Ysurac, 2023-04-29:
v2ray-plugin is only used to obfuscate ShadowSocks traffic. It's not related to v2ray proxy that will still be available.

kmrinal, 2023-04-30:
Ok, thank you for the clarification @Ysurac .

kmrinal, 2023-04-30:
@Kalimeiro You can proxy UDP traffic as well, if V2Ray is used for proxy.

Kalimeiro, 2023-04-30:
> @Kalimeiro You can proxy UDP traffic as well, if V2Ray is used for proxy.

I know, but it's bogus with 0.59.2 alpha.

closed 2023-04-30 by kmrinal