Ysurac/openmptcprouter #2839: Create Home Assistant Integration to manage OMR

JeedHome44, 2023-04-30:
Hello,

I use OMR and it's work perfectly but i want to see in Home Assistant if WAN1 is ok, WAN2 is ok, VPS is ok...
You see ? 

If someone can do an Home Assistant Integration to OMR, that will be so great !

Thank you !

falahati, 2023-08-16:
I also like to see the Wan status and wan bandwidth in HA. Altho I don't know if this is the right place to ask for this. but just in case I am adding my +1 to this issue as well.

spacesuitdiver, 2023-10-12:
@JeedHome44 @falahati are you running HA on the same device as OMR?

JeedHome44, 2023-10-12:
> @JeedHome44 @falahati are you running HA on the same device as OMR?

No 1 Raspberry for HA and 1 for OMR.

falahati, 2023-10-12:
My setup is a little complex, they are on the same hardware but on different virtual machines and have their own separate network interfaces. 

From the HA point of view, OMR is accessible on a different subnet than the rest of the network, behind a Router (x.x.1.1 - main network) and an instance of OpenWrt used for VPN (x.x.2.1) to finally reach the OMR at x.x.3.1.

But for almost all practical purposes, you can consider them separate devices.

I had some limited success with `SNMP` btw if that helps.

Ysurac, 2025-05-16:
WAN status should be possible to get in HA via SNMP (available in packages now, no more by default).
Else there is a local API with all the information, but not sure a HA module can already use them...