Ysurac/openmptcprouter #2844: configuration and install function ipsec site to site

Assistenzasystemistait, 2023-05-11:
Good morning,

Please I would like some help to install and configure the IPSEC site to site functionality, I tried to install some modules but it doesn't work.

Many thanks in advance

## Expected Behavior

## Current Behavior

## Specifications

  - OpenMPTCProuter version: 
  - OpenMPTCProuter VPS version: 
  - OpenMPTCProuter VPS provider:
  - OpenMPTCProuter platform:

Ysurac, 2023-05-16:
This is not related to OpenMPTCProuter.

Assistenzasystemistait, 2023-05-17:
I'm sorry, but it's not a feature present in openmtcprouter or did I have the wrong way to open tickets?
Thank you

Ysurac, 2023-05-17:
OpenMPTCProuter is based on OpenWRT, so IPsec is available but OpenMPTCProuter is made for connections aggregations. 
There is no native IPsec support on OpenMPTCProuter and you will have to refer to OpenWRT documentation.