Ysurac/openmptcprouter #2845: OMR-Bypass sometimes fails

sp2003, 2023-05-11:
## Expected Behavior
Sometimes, OMR-Bypass doesn't seem to work properly. I notice it from Netflix or some sites that don't open due to VPN, so I'm forced to restart OMR. Sometimes I have to turn off completely.
Is there a command from the CLI that can restart OMR-bypass for me?

## Current Behavior

## Specifications

  - OpenMPTCProuter version: turn off completely
  - OpenMPTCProuter VPS version: 5.4.207-mptcp 0.1028
  - OpenMPTCProuter VPS provider:
  - OpenMPTCProuter platform: VM Proxmox

Ysurac, 2023-05-16:
Check that you don't have another DHCP server or if another DNS than OpenMPTCProuter IP is not used on client.

sp2003, 2023-05-16:
hi

OMR is the only DHCP server.
It often happens on my PC which has a static IP and DNS and IPV6 disabled .
Actually, I haven't tested the other devices when the problem occurs.
Anyway, is it possible to restart OMR-bypass from CLI?

kevinh-csalabs, 2023-05-31:
Are you bypassing both Netflix ASNs 40027 and 2906?

sp2003, 2023-05-31:
certainly yes, but the problem is not Netflix.
Today, for example, I can't bypass real-debrid, I added the domain real-debrid.com and the asn AS212238 and also 212238.

kevinh-csalabs, 2023-05-31:
I'd confirm ipv6 isn't still responding as that definitely doesn't work properly. Try running tcptraceroute6 and tcptraceroute to see where they are going and confirm it is passing through the vps. You can also make sure ipv6 is disabled on the client in case something else is advertising those routes.

yostyle, 2023-06-26:
I have the same problem on my network. Sometimes OMR bypass doesn't work properly. As @sp2003 wrote I have to restart OpenMPTCP router to resolve this issue.