Ysurac/openmptcprouter #2848: Set a portion of bandwidth that cannot be touched for the Master interface

darthclide, 2023-05-28:
**Is your feature request related to a problem? Please describe.**
When saturating all connections, the master interface can briefly lose communication with the VPS which can result in slow webpage loading, or random disconnects in a video game

**Describe the solution you'd like**
If you could make it so that you can set a specific amount of bandwidth for the master interface to communicate with VPS, then this would not happen.

**Describe alternatives you've considered**
For now I have always just set an SQM that limits the upload and download to 15% less than the maximum bandwidth my provider offers. But this only alleviates the problem, not get rid of it entirely.

Ysurac, 2023-05-29:
It's what should do new SQM autorate on snapshot releases.

darthclide, 2023-05-29:
[image]
Are you talking about this? Are there any major downsides to using autorate? How would I go about filling the "minimum" and "maximum" fields to accomplish what I want?

Ysurac, 2023-05-30:
Hi, you can post here but I will also need the release used for that as the version in develop branch/snapshot is the same.

darthclide, 2023-05-31:
> Hy @Ysurac and @darthclide I´ve done some tests on the last weekend, and i have some results and thoughts about sqm autorate. Could i post here or is it better to open a new topic? Thanks.

Anything that can dedicate a set amount of bandwidth to the master interface, just let me know how to set those min/max thresholds.

darthclide, 2023-05-31:
My master connection is a solid DSL connection. Unfortunately there is nothing in that link except vague graphs, when I what I need is a simple "here is how you ensure some bandwidth is available for your master connection to communicate with the VPS"

darthclide, 2023-06-01:
Still hoping for some information on what numbers to try putting into "minimum/maximum" "download/upload" fields.

Ysurac, 2023-06-01:
In current release, you need to set the values that are working for the connection, sqm-autorate is not working correctly.
You need to lower values until you don't have issues anymore.

darthclide, 2023-06-01:
Do you have any snapshots available yet that I could try? And unfortunately lowering values does not seem to help sadly.

closed 2023-09-05 by github-actions[bot]