Ysurac/openmptcprouter #2852: Master connection not being used to ping VPS

darthclide, 2023-06-01:
## Expected Behavior
When pinging the VPS either from a PC in the network, or from the OMR router, it should use the Master connection (in my case DSL on eth1) to do so.

## Current Behavior
It uses my eth2 connection.

## Steps to Reproduce the Problem

  1. Disable all connections one by one. Each time I disable them and do a ping test, it uses another WAN but not my master (eth1)
  2. After I re-enable all connections, it does not matter which one was enabled last. It goes back to using my eth2 connection for pings.

## Context (Environment)
I believe it is what causes my OMR-bypass to keep restarting all the time (like once every 2-3 hours or sometimes every 30 minutes). As well as what causes it to show errors on status page such as "can't contact admin script" (even if nothing shows up in my system log)

## Specifications

  - OpenMPTCProuter version: Version 0.59.1-5.4
  - OpenMPTCProuter VPS version: Version 0.1028 5.4.207-mptcp
  - OpenMPTCProuter VPS provider: Vultr
  - OpenMPTCProuter platform: RPI4
  - Country: USA

darthclide, 2023-06-03:
Some extra info: My Verizon connection has always been shaky with 2 bars, but can reach 20 Mbps download at times so I keep it. But as a test, I decided to disable it. And OMR-bypass went from restarting every hour to only 1 or 2 times a day. There are 0 bypass rules that use that interface. And it is not set as my master. So why would OMR-bypass be restarting from 1 flaky connection? Unless it was somehow being used to communicate with the VPS, which it shouldn't since my DSL is set as my master.

Ysurac, 2023-06-05:
What is the result of ```ip r``` via SSH on the router ?

darthclide, 2023-06-05:
default via 10.255.255.1 dev tun0
default via 192.168.10.1 dev eth1 metric 3
default via 192.168.42.129 dev usb0 metric 10
default via 172.20.10.2 dev eth3 metric 14
default via 10.255.255.1 dev tun0 metric 1200
10.255.255.1 dev tun0 proto kernel scope link src 10.255.255.2
10.255.255.2 dev tun0 scope link metric 1200
45.63.64.60 via 192.168.42.129 dev usb0 metric 10
45.63.64.60 via 172.20.10.2 dev eth3 metric 14
127.0.0.0/8 dev lo proto static scope link metric 1
172.20.10.0/24 dev eth3 scope link metric 14
192.168.10.0/24 dev eth1 scope link metric 3
192.168.42.0/24 dev usb0 scope link metric 10

192.168.10.0 is my DSL and is the master connection.

Ysurac, 2023-06-05:
What do you have in System->OpenMPTCProuter, "Advanced settings" tab, at "Master interface selection" ?

darthclide, 2023-06-05:
On wizard change.

darthclide, 2023-06-05:
I can't really do that because I can't always find out the IP addresses or hostnames used to put into OMR-bypass. Aka, I don't want to be playing a game and have it suddenly start using my t-mobile connection because it has randomly become the Master. :/

darthclide, 2023-06-07:
Considering you are going to deprecate it, what will be my options for video games that I can't find the ports or domains used? I am currently leaving it in balancing mode per your advice.

But another consideration is that EAC (Easy Anti-Cheat) is very sensitive to network changes. Even when I am on my DSL connection, even a short 2 second dropout (rare) is enough to kick me out of the game. So even if balancing mode results in 0 ping spikes, getting kicked out of a game is no fun.

Ysurac, 2023-06-07:
In any case connection with lower latency have an higher priority.

darthclide, 2023-06-07:
So you are saying that on balance mode it will never switch off of DSL unless it can't ping server on it?

Ysurac, 2023-06-08:
Connections are used based on latency, lower latency first.
So others connections will be also used if needed (when you need higher bandwidth or when connection is down), but it's already the case in your mode.

closed 2023-09-12 by github-actions[bot]

ioogithub, 2023-11-05:
> Well after 2 days of being on balance mode, I haven't seen OMR-bypass restart. And none of my games have disconnected. 

Did you ever see omr switch the master? Mine has always been on balancing. Currently one wan is struggling with high pings and very high latency, omr never switches over to the other wan.

[Ysurac](https://github.com/Ysurac) Does this setting work? I have never seen it switch once in over 2 years working with OMR. Should I try the other modes such as Dynamic Change?

What script is responsible for monitoring this and under what conditions does it change?

Ysurac, 2023-11-05:
Old issue...
OMR never switch master while master is working in balancing mode. This is not needed to change it.
OMR-Tracker is the script that detect when connection doesn't work.

ioogithub, 2023-11-05:
> Old issue... OMR never switch master while master is working in balancing mode. This is not needed to change it.

What does this setting do then? The options are:

    On Wizard Change
    Dynamic Change
    No Change
    Balancing

Does dynamic change, change the master or does this setting not do anything? What does balancing actually do, how is it different than one wizard change then?

Ysurac, 2023-11-05:
Dynamic change master based on latency on the fly, balancing use all connections in a load balancing mode. On wizard change mode only change master when wizard page is saved.
You should keep "Balancing mode", not sure if I will keep others modes...

darthclide, 2023-11-05:
> Did you ever see omr switch the master? Mine has always been on balancing. Currently one wan is struggling with high pings and very high latency, omr never switches over to the other wan.

I have never seen my Master change. But from what I gather, you are supposed to ignore the Master interface when on balance mode, and focus more on if you are getting laggy websites combined with errors in your log. Otherwise you can ignore what interface is Master.

Though it does still make me wonder why the word "Master" is still used, if the software correctly abandons using a wan for communicating with the VPS as soon as the ping gets too high.

Ysurac, 2023-11-05:
Master connection is used for initial connection only. It's also a connection with higher priority when VPS connection is lost.

darthclide, 2023-11-06:
> Master connection is used for initial connection only. It's also a connection with higher priority when VPS connection is lost.

Yeah, I just don't think the algorithms are quite there yet with seamless switching between two WANS communicating with VPS.

The primary reason I haven't had problems starting 3 months ago is that Verizon built a new tower from scratch that has 5G on it. And this tower is 500 feet from me. This means all my 4G connections average 50ms, which means even if my DSL goes down, my system log page is still only filled with normal udhcpc and dnsmasq messages.

ioogithub, 2023-11-06:
> Master connection is used for initial connection only.

Is this true? I just did another test, wan1 is the master. Simulate power going out on wan1 but not wan2. OMR shows wan2 is green,everything else is orange: VPN is down, v2ray is down, can't contact VPS. Whole system is down, no internet yet wan2 is ready to be used.

The tracker and post tracker script keep running, adding routes, removing routes, but no internet until power is restored to wan1. There is no redundancy, the script can't recover from the master being down. I waited 30m, still no internet and it never even failed over to direct connection on wan2.

 It should failover to wan1 but it does not. The issue is also outlined here: https://github.com/Ysurac/openmptcprouter/issues/2996. How can I get omr to improve the fail-over capabilities? It really doesn't seem to work.

Ysurac, 2023-11-06:
It's not related to initial issue of this closed issue.