Ysurac/openmptcprouter #2854: Proxy shadowsocks fails down

adrirka, 2023-06-11:
## Expected Behavior
I try to understand why the proxy shadowsock fails down right now without any actions setting from my side (OMR/VPS), then everything was working well until this day.

## Current Behavior
The system log displays the error message : "daemon.err /usr/bin/ss-local[13543]: getpeername: Socket not connected"
Please seeing the screenshot below os statut configuration : .

[image]

## Possible Solution
I suppose it due to the kernel version which has switched in the bootloader .

-------------------------------------------------------
root@vps-de78803b:/home/debian# ls /boot
System.map-5.10.0-22-cloud-amd64  grub
System.map-5.10.0-23-cloud-amd64  initrd.img-5.10.0-22-cloud-amd64
System.map-5.4.207-mptcp          initrd.img-5.10.0-23-cloud-amd64
config-5.10.0-22-cloud-amd64      initrd.img-5.4.207-mptcp
config-5.10.0-23-cloud-amd64      vmlinuz-5.10.0-22-cloud-amd64
config-5.4.207-mptcp              vmlinuz-5.10.0-23-cloud-amd64
efi                               vmlinuz-5.4.207-mptcp
-------------------------------------------------------

## Context (Environment)
I didn't update the kernel or install any new software on the VPS instance or the OMR server.
The shadow key is still setup

## Specifications
  - OpenMPTCProuter version: 
  - OpenMPTCProuter VPS version: Debian 11
  - OpenMPTCProuter VPS provider: OVH
  - OpenMPTCProuter platform: RPI3B+
  - Country: FR

=========================================================

Is there any setting should i do through OMR to recover the boot on mptcp kernel version 5.4.207-mptcp ? 
Or a solution to fix it with the VM instance ?
An then if you have any idea to avoid the auto switch like that, i'm interested in : )

Many thanks for helping !!

Ysurac, 2023-06-12:
You can try to launch again VPS install script.

adrirka, 2023-06-16:
Yes OK sounds good. 
Thanks !