Ysurac/openmptcprouter #2856: Setting up Wi-Fi for the main network

anti-spy, 2023-06-16:
Hello. I want to share a local network from the built-in RPI 4b Wi-Fi. So that the connected devices are connected via DHСP 192.168.100.~. I found the official instructions, but they suggest creating a different subnet. How to do it?

## Specifications

  - OpenMPTCProuter version: v0.59.1-5.4 r0+16594-ce92de8c8c
  - OpenMPTCProuter VPS version: 
  - OpenMPTCProuter VPS provider: itldc.com
  - OpenMPTCProuter platform: RPI4B

gizze59, 2023-06-18:
This work out-of-the box on my OpenMPTCProuter running same image as you. 
What I did. 
First I Removed the default OpenWRT wireless interface and Save-Apply.
Under Status-Channel Analysis I choose Radio0(5ghz) and find channel 36 was best in my area for 40Mhz
Under Wireless "radio0" choose Add
Under Operating frequency Mode: AC , Channel 36 , Width: 40Mhz
Under General setup, Mode Access Point, Your ESSID (default OpenWrt), NetWork: lan
Under Wireless security, Encryption: WPA2-PSK, Key: yourchoice, Save&Apply
When you open Network-Interfaces then you will be asked to enable bridge mode for lan

Hope this help
Here is my /etc/config/wireless
config wifi-device 'radio0'
	option type 'mac80211'
	option path 'platform/soc/fe300000.mmcnr/mmc_host/mmc1/mmc1:0001/mmc1:0001:1'
	option channel '36'
	option band '5g'
	option cell_density '0'
	option htmode 'VHT40'

config wifi-iface 'wifinet0'
	option device 'radio0'
	option mode 'ap'
	option ssid 'OpenWrt'
	option network 'lan'
	option encryption 'psk2'
	option key 'mysecretpassword'

anti-spy, 2023-08-28:
It's not stable for me.  After booting the system, the access point does not turn on.  If you turn it off and on, it sometimes starts to work.  Sometimes a device connects to an access point, but the device does not have access to the local network.  In general, this does not work at all satisfactorily.

anti-spy, 2023-08-28:
esp-roof-of-the-house | ---/-76 дБм | 0.0 Мбит/с, 0 МГц1.0 Мбит/с, 20 МГц

the device is connected, but the reception shows 0 and it is not on the local network. At least the bridge is up and running.

gizze59, 2023-09-03:
Could you share /etc/config/wireless or "uci show wireless" ?

anti-spy, 2023-09-06:
Now the network does not rise at all, it immediately turns off.  A couple of days everything worked, after rebooting it does not work.

`
root@OpenMPTCProuter:~# cat /etc/config/wireless

config wifi-device 'radio0'
        option type 'mac80211'
        option path 'platform/soc/fe300000.mmcnr/mmc_host/mmc1/mmc1:0001/mmc1:0001:1'
        option channel 'auto'
        option country 'LV'
        option cell_density '1'
        option band '2g'

config wifi-iface 'wifinet0'
        option device 'radio0'
        option mode 'ap'
        option ssid 'VIP NETWORK'
        option encryption 'psk2'
        option key 'password'
        option wpa_disable_eapol_key_retries '1'
        option network 'lan'

ifconfig -a :

wlan0     Link encap:Ethernet  HWaddr DC:A6:32:F3:5C:78
          BROADCAST PROMISC ALLMULTI MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

root@OpenMPTCProuter:~# iw wlan0 info
Interface wlan0
        ifindex 13
        wdev 0x1
        addr dc:a6:32:f3:5c:78
        type managed
        wiphy 0
        channel 36 (5180 MHz), width: 20 MHz, center1: 5180 MHz

root@OpenMPTCProuter:~# iw list
Wiphy phy0
        wiphy index: 0
        max # scan SSIDs: 10
        max scan IEs length: 2048 bytes
        max # sched scan SSIDs: 16
        max # match sets: 16
        Retry short limit: 7
        Retry long limit: 4
        Coverage class: 0 (up to 0m)
        Available Antennas: TX 0 RX 0
        Supported interface modes:
                 * IBSS
                 * managed
                 * AP
                 * P2P-client
                 * P2P-GO
                 * P2P-device
        Band 1:
                Capabilities: 0x1022
                        HT20/HT40
                        Static SM Power Save
                        RX HT20 SGI
                        No RX STBC
                        Max AMSDU length: 3839 bytes
                        DSSS/CCK HT40
                Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
                Minimum RX AMPDU time spacing: 16 usec (0x07)
                HT TX/RX MCS rate indexes supported: 0-7
                Frequencies:
[... 71 lines cut ...]
                        * 5640 MHz [128] (20.0 dBm) (radar detection)
                        * 5660 MHz [132] (20.0 dBm) (radar detection)
                        * 5680 MHz [136] (20.0 dBm) (radar detection)
                        * 5700 MHz [140] (20.0 dBm) (radar detection)
                        * 5720 MHz [144] (13.0 dBm) (radar detection)
                        * 5745 MHz [149] (13.0 dBm)
                        * 5765 MHz [153] (13.0 dBm)
                        * 5785 MHz [157] (13.0 dBm)
                        * 5805 MHz [161] (13.0 dBm)
                        * 5825 MHz [165] (13.0 dBm)
        valid interface combinations:
                 * #{ managed } <= 1, #{ P2P-device } <= 1, #{ P2P-client, P2P-GO } <= 1,
                   total <= 3, #channels <= 2
                 * #{ managed } <= 1, #{ AP } <= 1, #{ P2P-client } <= 1, #{ P2P-device } <= 1,
                   total <= 4, #channels <= 1
        max # scan plans: 1
        max scan plan interval: 508
        max scan plan iterations: 0
        Supported extended features:
                * [ 4WAY_HANDSHAKE_STA_PSK ]: 4-way handshake with PSK in station mode
                * [ 4WAY_HANDSHAKE_STA_1X ]: 4-way handshake with 802.1X in station mode
                * [ DFS_OFFLOAD ]: DFS offload
                * [ SAE_OFFLOAD ]: SAE offload support
                * [ 4WAY_HANDSHAKE_AP_PSK ]: AP mode PSK offload support
`

anti-spy, 2023-09-06:
how to turn off wlan after manual restart   [https://youtu.be/gf8gxnOZcO8?si=GO13t-2gqBQ3_zYQ ](url)

closed 2023-12-10 by github-actions[bot]