Ysurac/openmptcprouter #2861: rpi4 custom image build error

bluisana, 2023-06-26:
## Expected Behavior
I am trying to build a custom image for rpi4. 

This is the build command I am using. 

OMR_TARGET="rpi4" OMR_FEED_SRC="v0.59.2alpha"

## Current Behavior
The build fails here. 

curl -4 -f --connect-timeout 20 --retry 5 --location --insecure https://mirror2.openwrt.org/sources/ndisc6_1.0.3.orig.tar.bz2

curl: (22) The requested URL returned error: 404 Not Found
Download failed.
No more mirrors to try - giving up.

Ysurac, 2023-06-27:
This is fixed in develop branch.

bluisana, 2023-06-27:
Thanks Ysurac, 

Now I get an error about file changes in my branch when trying to build but when I do a git diff it doesn't show any changes.  I am assuming that the build script in develop is checking out a different branch?  How do I fix this? 

$git checkout develop
$OMR_TARGET="rpi4" OMR_FEED_SRC="develop" ./build.sh V=s
error: Your local changes to the following files would be overwritten by checkout:
        include/kernel-version.mk
        include/netfilter.mk
        package/base-files/files/bin/config_generate
        package/kernel/linux/modules/crypto.mk
        package/kernel/linux/modules/fs.mk
        package/kernel/linux/modules/netfilter.mk
        package/kernel/linux/modules/netsupport.mk
        package/kernel/linux/modules/other.mk
        scripts/download.pl
        target/linux/generic/config-5.4
        target/linux/generic/pending-5.4/640-netfilter-nf_flow_table-add-hardware-offload-support.patch
        target/linux/generic/pending-5.4/645-netfilter-nf_flow_table-rework-hardware-offload-time.patch
Please commit your changes or stash them before you switch branches.
Aborting

Ysurac, 2023-06-27:
```rm -rf rpi4```

bluisana, 2023-06-28:
I finally got everything to compile but there is just a packages and targets folder in my source/bin folder.  Is this supposed to create the full image file or do I have to take some further manual steps to do that?  

One thing I noticed was that I have quite a few folders with dump.txt files in the logs directory.  These seem to be associated with package or dependency errors.  Are these normal to run into during a build?

Ysurac, 2023-06-28:
Images should be in rpi/6.1/bin/targets/bcm27xx/bcm2711/ if compilation work successfully.
dump.txt are normal.

Ysurac, 2023-06-28:
you need to add ```OMR_KERNEL="6.1"``` to the command line.

bluisana, 2023-06-29:
Thanks, 

This image finally recognized all of the ethernet adapters on my PI :).  The testing will begin shortly :).

closed 2023-10-03 by github-actions[bot]