Ysurac/openmptcprouter #2862: Installation OMR KERNEL 6.1

Henrb, 2023-06-27:
Bonsoir,

Je tente d'installer OMR avec le Kernel 6.1 mais je ne parviens pas a faire fonctionner la VM, de l'aide serait la bien venu.
Voici l'erreur :
[image]

Ysurac, 2023-06-27:
Il n'y a pas vraiment d'erreur là. En pressant "Entrée" ça affiche quelque chose ?
C'est quelle image exactement ?

Henrb, 2023-06-27:
> Il n'y a pas vraiment d'erreur là. En pressant "Entrée" ça affiche quelque chose ? C'est quelle image exactement ?

Désoler je ne vois vraiment pas comment me débloquer, le shell me dit busy box built-in shell ? Une idée ? 
Merci par avance

Henrb, 2023-06-27:
Re, j'ai avancer un peu je suis bloquer ici maintenant IMG_2963.HEIC

Ysurac, 2023-06-28:
Il y a un souci sur la dernière image, je travaille dessus.

Henrb, 2023-06-28:
Ok, merci pour le retour. Tiens nous au courant, pour la version la : http://www.openmptcprouter.com/download il y a le package .ova, serait il possible de l'avoir sur la version 6.1 ?

Kalimeiro, 2023-07-02:
> 

Salut, as-tu fait un essai pour vérifier que l'agrégation se fait bien avec le noyau 6.1, car dans mes précédents essais, c'était négatif, il n'y avait que la connexion "master" qui fonctionnait, les autres liens n'étaient pas agrégés.

closed 2023-10-30 by github-actions[bot]