Ysurac/openmptcprouter #2864: Remote access to openmptcp router

mvmaksnet, 2023-06-27:
Hello.
What I need to do on rpi3 openmptcp router so can manage it from remote location?
Example:
Router is on the field with camerman and encoder and I am managing router from my office.

Ysurac, 2023-06-29:
You can access router part from VPS, you need to allow ports via web interface in firewall configuration.

mvmaksnet, 2023-06-29:
I know that, but how? I know how to do it on mikrotik, but here is different.

gizze59, 2023-06-29:
This depend of which kind of control you have on your VPS. If you like me, run your own VPS you can do SSH from VPS to openmptcp router, either to tun0 interface (default 10.255.255.2) or to router LAN ip (default 192.168.100.1). 
You need to change Firewall rules for Zones to accept on Input (default reject) and add a Traffic rule allow SSH (tcp/22) from source zone "vpn" to destination zone vpn og lan. I using lan, but then you need to add a static route on VPS to send det network 192.168.100.0/24 via tun0 ip (default 10.255.255.2)
You can do firewall rules either in /etc/config/firewall directly or using UCI command, but UI is more easy :-)

closed 2023-10-03 by github-actions[bot]