Ysurac/openmptcprouter #2866: where is the sysytemlog in 0.59.2 ?how to checkout it.

fareign, 2023-06-29:
## Expected Behavior

## Current Behavior

 In Status->System log is disappear. 
the chages log  Config :Use Busybox Syslogd instead of OpenWRT logd
but I cant find it.

## Specifications

  - OpenMPTCProuter version: 0.59.2
  - OpenMPTCProuter VPS version: 0.1028
  - OpenMPTCProuter VPS provider:aliyun
  - OpenMPTCProuter platform: x86_64

Ysurac, 2023-06-29:
There is no 0.59.2 release yet. If you mean snapshot image, it's still an alpha release, changes are made every days and it's not yet fully tested.

fareign, 2023-06-29:
> There is no 0.59.2 release yet. If you mean snapshot image, it's still an alpha release, changes are made every days and it's not yet fully tested.

I know is alpha ,but I want to see the syslog and I cant find it .help me.

Ysurac, 2023-06-29:
You can use ```logread``` command via SSH.
This will be fixed in web interface, system log menu, in next image build.

vempire-ghost, 2023-06-30:
Can you fix the logs in the 5.4 kernel too pls?

Kalimeiro, 2023-07-02:
It's fixed since latest snapshot.

vempire-ghost, 2023-07-05:
[image]
Its still broken in the 5.4 kernel, they only fixed in the 6.1.

Kalimeiro, 2023-07-05:
[image]

vempire-ghost, 2023-07-06:
How strange, I'm using the x86_x64 image in VirtualBox, are you using it too?

Kalimeiro, 2023-07-06:
I'm using x86_64 image on a local dedicated server (Ryzen 5 5600G + SFP+ x 2 pci-e card + 2.5Gbps x 4 pci-e card).
I'm always using latest snapshot when available for help to debug.
I thinks your problem is somewhere in virtualbox.

vempire-ghost, 2023-07-06:
Indeed, the snapshots for VirtualBox with kernel 6.1 have the system log working, but the ones with kernel 5.4 don't. It may be something specific to this image that is still bugged. If Ysurac could take a look, I would appreciate it. I would like to go back to using the latest versions for testing, but it's difficult without the log.

Ysurac, 2023-07-06:
It's working also with 5.4 kernel.

closed 2023-10-09 by github-actions[bot]