Ysurac/openmptcprouter #2868: OMR 6.1 cannot connect with VPS.

vempire-ghost, 2023-07-01:
## Expected Behavior

OMR connects with the VPS.
## Current Behavior

Jul  1 01:15:26 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later (can ping server vps on x.x.x.x, no server API answer on x.x.x.x)

OMR can ping but cannot retrieve keys from de VPS, in VPS when a run uname -a show this Linux debian-94311 6.1.0-9-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.27-1 (2023-05-08) x86_64 GNU/Linux.

Ports are open and OMR 5.4 works great in my second VPS running in this provider, only the 6.1 cannot retrieve keys.
I used this script in the VPS wget -O - http://www.openmptcprouter.com/server-test/debian.sh | UPSTREAM6="yes" sh

## Specifications

  - OpenMPTCProuter version: v0.59.2alpha-6.1 r0+23414-ed816f6ba8
  - OpenMPTCProuter VPS version:  OpenMPTCProuter VPS 0.1029-test 
  - OpenMPTCProuter VPS provider: Oracle Cloud
  - OpenMPTCProuter platform: Virtual Box x86 x64

aliosa27, 2023-07-10:
if you run netstat -ln , do you see port 65000 listed? if not, it seems that python's version was updated and that causes an issue with some regex functionality that causes the admin server to fail. manual configuration works

vempire-ghost, 2023-07-10:
There is some ports above 65001 but not one with 650000 exactly.
What kind of manual configuration works?

closed 2023-10-14 by github-actions[bot]