Ysurac/openmptcprouter #2870: How to improve UDP performance on OpenMPTCPRouter

MXNPD, 2023-07-01:
## Expected Behavior
Expected to get more than UDP speed even when a connection is capable of more than what is being given.

## Current Behavior
UDP speed is lower than expected. I have a test running in a data center environment in Dallas TX with a VPS from Vultr and my own dedicated server. I set up 3 pfsense VM's each with their own public IP address. I then fed the LAN of each of those into their own port profile in ESXi, with each one being limited to 350mbps, my data center has a 1Gbps connection for the dedicated server. I then brought those into OpenMPTCPRouter and setup a windows client on the LAN interface of OpenMPTCPRouter. I also setup the VPS end on Debian 11 on Vultr with the VPS Script. With TCP, the all 3 connections are maxed out and I get 900+mbps up/down. For this test I also used the proxy V2Ray in the settings  UDP speed tests are not very good.  I was trying to simulate what it would be like if I had one of my VPN apps enabled on a client, or one of my wireguard S2S connections. It works, just not very well. I set up a isolated wireguard server myself (also in Vultr, separate VPS) along with using PIA (Private Internet Access) using their wireguard backend. Given that wireguard is UDP based, the results were about much less than what I would like them to be. I get about 130mbps down and about 80 mbps up (on both PIA and my wireguard server test). 

Is there a way that UDP performance could be improved? I would love to use OpenMPTCPRouter but I have so many things that sadly rely on UDP traffic. 

## Specifications

  - OpenMPTCProuter version:  openmptcprouter-v0.59.1-5.4-r0+16594-ce92de8c8c
  - OpenMPTCProuter VPS version:  Latest 5.4 kernel version, ran script 3 days ago, did not see direct version number
  - OpenMPTCProuter VPS provider: Vultr
  - OpenMPTCProuter platform:  x86_64

simonemessina92, 2023-07-25:
I mainly would like to use OMR for UDP and SRT transmission but sadly speaking didn't have found a proper setting to make it work properly. I think we shall need a fully functional v1 realease sooner or later. I 've found many problem using WebRTC and SRT togheter, the only setting worked properly on one of the custom multi lan pc I got with OMR running the 0.59.2 beta is to settle V2ray and no VPN at all ... It's like proxy block VPN and viceversa with some UDP traffic design applications, for example vMix call won't work with TCP bonding alghorithm in the proxy, this software is still a big equation that need some more fine tuning unfortunately; but I've got faith we can see something better in the future!

Ysurac, 2023-07-26:
UDP performance should be better in latest snapshot image.

closed 2023-10-30 by github-actions[bot]