Ysurac/openmptcprouter #2874: cake-autorate.omrvpn: ERROR

fareign, 2023-07-06:
## Expected Behavior
Suddenly unable to use multipath，only one path.
I think there is some thing relate to cake-autorate.

## Current Behavior
Jul  6 00:58:33 OpenMPTCProuter user.notice post-tracking-post-tracking: Interface route not yet set, set route ip r add default via 192.168.220.1 dev usb1 metric 4
Jul  6 01:00:00 OpenMPTCProuter user.notice cake-autorate.omrvpn: INFO: 1688605200.540412 Warning: Configured global ping response timeout: 100.0 seconds exceeded.
Jul  6 01:01:21 OpenMPTCProuter user.notice cake-autorate.omrvpn: ERROR: 1688605281.268002 Received request to change maintain_pingers state to existing state.
Jul  6 01:03:54 OpenMPTCProuter user.notice cake-autorate.omrvpn: INFO: 1688605434.773045 Warning: Configured global ping response timeout: 100.0 seconds exceeded.

Jul  6 01:10:25 OpenMPTCProuter user.notice cake-autorate.omrvpn: INFO: 1688605825.223718 Warning: Configured global ping response timeout: 100.0 seconds exceeded.
Jul  6 01:11:56 OpenMPTCProuter user.notice cake-autorate.omrvpn: ERROR: 1688605916.070937 Received request to change maintain_pingers state to existing state.
Jul  6 01:13:35 OpenMPTCProuter user.notice cake-autorate.omrvpn: INFO: 1688606015.739374 Warning: Configured global ping response timeout: 100.0 seconds exceeded.
Jul  6 01:15:18 OpenMPTCProuter user.notice cake-autorate.omrvpn: ERROR: 1688606118.329550 Received request to change maintain_pingers state to existing state.
Jul  6 01:17:31 OpenMPTCProuter user.notice post-tracking-post-tracking: wan5 (usb4) switched off because check error and ping from 192.168.223.37 error (9.9.9.9,1.0.0.1,114.114.115.115)
Jul  6 01:17:31 OpenMPTCProuter user.notice post-tracking-post-tracking: Delete default route to 112.74.50.106 via 192.168.223.1 dev usb4
Jul  6 01:17:44 OpenMPTCProuter user.notice post-tracking-post-tracking: wan5 (usb4) switched up
Jul  6 01:17:44 OpenMPTCProuter user.notice post-tracking-post-tracking: Interface route not yet set, set route ip r add default via 192.168.223.1 dev usb4 metric 11
Jul  6 01:18:18 OpenMPTCProuter user.notice post-tracking-post-tracking: wan5 (usb4) switched off because check error and ping from 192.168.223.37 error (1.1.1.1,4.2.2.1,8.8.8.8)
Jul  6 01:18:18 OpenMPTCProuter user.notice post-tracking-post-tracking: Delete default route to 112.74.50.106 via 192.168.223.1 dev usb4
Jul  6 01:18:25 OpenMPTCProuter user.notice cake-autorate.omrvpn: INFO: 1688606305.532189 Warning: Configured global ping response timeout: 100.0 seconds exceeded.
Jul  6 01:18:30 OpenMPTCProuter user.notice post-tracking-post-tracking: wan5 (usb4) switched up
Jul  6 01:18:30 OpenMPTCProuter user.notice post-tracking-post-tracking: Interface route not yet set, set route ip r add default via 192.168.223.1 dev usb4 metric 11

## Possible Solution

## Steps to Reproduce the Problem

  1.user 5 simcard,and 4 card is in multipath
  2.push flow is 6mb/s
  3.after 25minitues, only one path is in use.
4.stop push,
5.push again , works well

## Context (Environment)

## Specifications

  - OpenMPTCProuter version: 0.59.2alpha
  - OpenMPTCProuter VPS version: 0.1028
  - OpenMPTCProuter VPS provider:
  - OpenMPTCProuter platform: x86_64
  - Country: CN

fareign, 2023-07-06:
socket:ss
VPN：glorytunUDP,
MPTCP Settings: fullmesh,BLEST,bbr

Ysurac, 2023-07-06:
Did you set speed for each interface in the wizard ?

fareign, 2023-07-06:
> Did you set speed for each interface in the wizard ?

no

fareign, 2023-07-07:
If I check  interface's SQM instance. then below err log. (VPS:0.1029 test 5.4 )

`Jul  7 17:35:49 OpenMPTCProuter user.notice cake-autorate.omrvpn: ERROR: 1688751349.785396 /usr/share/sqm-autorate/cake-autorate.sh: line 2068: achieved_rate_kbps[ul]: unbound variable
Jul  7 17:35:49 OpenMPTCProuter user.notice cake-autorate.omrvpn: ERROR: 1688751349.790321 /usr/share/sqm-autorate/cake-autorate.sh: line 1: pinger_pid: unbound variable
Jul  7 17:35:50 OpenMPTCProuter user.notice cake-autorate.omrvpn: INFO: 1688751350.990417 Stopped cake-autorate with PID: 16519 and config: /usr/share/sqm-autorate/cake-autorate_config.omrvpn.sh
Jul  7 17:35:54 OpenMPTCProuter user.err : Error reading total packet size: -1
Jul  7 17:35:54 OpenMPTCProuter user.err : Stoping client
Jul  7 17:36:01 OpenMPTCProuter user.notice sqm: latency omrvpn: 300
Jul  7 17:36:01 OpenMPTCProuter user.notice cake-autorate.omrvpn: INFO: 1688751361.052951 Starting cake-autorate with PID: 22722 and config: /usr/share/sqm-autorate/cake-autorate_config.omrvpn.sh
Jul  7 17:36:05 OpenMPTCProuter user.notice post-tracking-post-tracking: wan4 (usb3) switched up
Jul  7 17:36:05 OpenMPTCProuter user.notice post-tracking-post-tracking: Interface route not yet set, set route ip r add default via 192.168.223.1 dev usb3 metric 10
Jul  7 17:37:01 OpenMPTCProuter user.notice cake-autorate.omrvpn: ERROR: 1688751421.137714 /usr/share/sqm-autorate/cake-autorate.sh: line 2068: achieved_rate_kbps[ul]: unbound variable
Jul  7 17:37:01 OpenMPTCProuter user.notice cake-autorate.omrvpn: ERROR: 1688751421.142051 /usr/share/sqm-autorate/cake-autorate.sh: line 1: pinger_pid: unbound variable
Jul  7 17:37:02 OpenMPTCProuter user.notice cake-autorate.omrvpn: INFO: 1688751422.345073 Stopped cake-autorate with PID: 22722 and config: /usr/share/sqm-autorate/cake-autorate_config.omrvpn.sh
Jul  7 17:37:12 OpenMPTCProuter user.notice sqm: latency omrvpn: 300
Jul  7 17:37:12 OpenMPTCProuter user.notice cake-autorate.omrvpn: INFO: 1688751432.407560 Starting cake-autorate with PID: 2146 and config: /usr/share/sqm-autorate/cake-autorate_config.omrvpn.sh
Jul  7 17:38:12 OpenMPTCProuter user.notice cake-autorate.omrvpn: ERROR: 1688751492.495250 /usr/share/sqm-autorate/cake-autorate.sh: line 2068: achieved_rate_kbps[ul]: unbound variable
Jul  7 17:38:12 OpenMPTCProuter user.notice cake-autorate.omrvpn: ERROR: 1688751492.497140 /usr/share/sqm-autorate/cake-autorate.sh: line 1: pinger_pid: unbound variable
Jul  7 17:38:13 OpenMPTCProuter user.notice cake-autorate.omrvpn: INFO: 1688751493.702704 Stopped cake-autorate with PID: 2146 and config: /usr/share/sqm-autorate/cake-autorate_config.omrvpn.sh
Jul  7 17:38:23 OpenMPTCProuter user.notice sqm: latency omrvpn: 300
Jul  7 17:38:23 OpenMPTCProuter user.notice cake-autorate.omrvpn: INFO: 1688751503.765885 Starting cake-autorate with PID: 12629 and config: /usr/share/sqm-autorate/cake-autorate_config.omrvpn.sh`

closed 2023-10-11 by github-actions[bot]