Ysurac/openmptcprouter #2875: OMR Bypass via ASN with Firewalla in front of OMR?

bluisana, 2023-07-06:
I have a Firewalla device between my local devices and OMR.  I can't get rid of this device because I have all sorts of rules for childrens devices that can't be mimiced in OMR

## Expected Behavior
OMR rules work while there is a firewall between OMR and my local devices. 

## Current Behavior
OMR rules are ignored because my Firewalla device does DNS for my local networks. 

Is there any type of tagging or fancy DNS passthrough I can do to keep my Firewalla device on my network and use ASN bypass on my OMR?

Ysurac, 2023-07-07:
You should use Protocols & service and use Disneyplus here.

bluisana, 2023-07-07:
When I try this the dropdown for selecting what protocol/service I want to add is completely blank.

Ysurac, 2023-07-07:
What is the release used ?

bluisana, 2023-07-07:
[v0.59.2alpha-6.1 r0+23414-ed816f6ba8](https://github.com/ysurac/openmptcprouter)

Ysurac, 2023-07-09:
This should be fixed in latest snapshot of will be in next snapshot.

closed 2023-10-12 by github-actions[bot]