Ysurac/openmptcprouter #2876: DNS forwardings randomly loses settings

raidzx, 2023-07-08:
## Expected Behavior

When I set the DNS server on DHCP and DNS under DNS forwardings the server should persist until I change it
## Current Behavior

The DNS server randomly disappears and is replace with 127.0.0.1:5353. This seems to happen when I reboot the system and when interfaces have their status change

## Possible Solution

n/a
## Steps to Reproduce the Problem

  1. Add server in DNS Forwardings on DHCP and DNS page under "Network"
  2. Wait for server to disappear and be replaced with 127.0.0.1#5353
  3.

## Context (Environment)

I have another MPTCP router with similar configurations where this does not happen

## Specifications

  - OpenMPTCProuter version: 0.59.1
  - OpenMPTCProuter VPS version: 0.1028
  - OpenMPTCProuter VPS provider: XenSpec
  - OpenMPTCProuter platform: x86_64
  - Country: USA

Ysurac, 2023-07-10:
I'm not able to reproduce the issue.
Can you give me, via SSH on the router, the result of ```uci show dhcp``` when you made DNS change ? I would also need the result of ```uci get openmptcprouter.vps.pihole``` (to check if there is not a wrong setting here).

Ysurac, 2023-07-10:
I found the issue: this is due to using 10.2*#53 as DNS server. Remove the #53 (it use port 53 as default) and this will be solved.
I fix this for next release.

raidzx, 2023-07-10:
Wow, I am glad it is such a simple fix. Appreciate your help. #53 removed!

closed 2023-07-13 by Ysurac