Ysurac/openmptcprouter #2877: omr-admin.py error

qweryu342, 2023-07-09:
## Expected Behavior
OMPTCP Router contact server Admin Script

## Current Behavior
OMPTCP Router cant contact server Admin Script

In VPS  omr-admin.py is 100% cpu usage. And always restarting (PID changing) 

This is 

● omr-admin.service - OMR-Admin
     Loaded: loaded (/lib/systemd/system/omr-admin.service; disabled; vendor preset: enabled)
     Active: active (running) since Sun 2023-07-09 15:17:59 MSK; 3s ago
   Main PID: 61119 (python3)
      Tasks: 1 (limit: 620)
     Memory: 47.7M
     CGroup: /system.slice/omr-admin.service
             └─61119 python3 /usr/local/bin/omr-admin.py

Jul 09 15:18:02 p647538.kvmvps omr-admin.py[61119]: pydantic_core._pydantic_core.SchemaError: Error building "model" validator:
Jul 09 15:18:02 p647538.kvmvps omr-admin.py[61119]:   SchemaError: Error building "model-fields" validator:
Jul 09 15:18:02 p647538.kvmvps omr-admin.py[61119]:   SchemaError: Field "remoteip6":
Jul 09 15:18:02 p647538.kvmvps omr-admin.py[61119]:   SchemaError: Error building "default" validator:
Jul 09 15:18:02 p647538.kvmvps omr-admin.py[61119]:   SchemaError: Error building "nullable" validator:
Jul 09 15:18:02 p647538.kvmvps omr-admin.py[61119]:   SchemaError: Error building "str" validator:
Jul 09 15:18:02 p647538.kvmvps omr-admin.py[61119]:   SchemaError: regex parse error:
Jul 09 15:18:02 p647538.kvmvps omr-admin.py[61119]:     (?:^|(?<=\s))(([0-9a-fA-F]{1,4}:){7,7}[0-9a-fA-F]{1,4}|([0-9a-fA-F]{1,4}:){1,7}:|([0-9a-fA-F]{1,4}:){>
Jul 09 15:18:02 p647538.kvmvps omr-admin.py[61119]:          ^^^^
Jul 09 15:18:02 p647538.kvmvps omr-admin.py[61119]: error: look-around, including look-ahead and look-behind, is not supported

## Possible Solution

## Context (Environment)

## Specifications

  - OpenMPTCProuter version: v0.59.1-5.4 r0
  - OpenMPTCProuter VPS version: OMR_VERSION="0.1028"
  - OpenMPTCProuter VPS provider: kvm (debian and ubuntu all) 
  - OpenMPTCProuter platform: r2s (rockchip)
  - Country:

Ysurac, 2023-07-09:
This seems to be a problem on IPv6 detection.
What is the result of ``` ip -6 a``` and what is the VPS provider ?

qweryu342, 2023-07-09:
vps provider is specific (not from faq list): 

root@p647538:~# ip -6 a
1: lo: <LOOPBACK,UP,LOWER_UP,80000> mtu 65536 state UNKNOWN qlen 1000
    inet6 ::1/128 scope host
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 state UP qlen 1000
    inet6 fe80::200:ff:fe64:7538/64 scope link
       valid_lft forever preferred_lft forever
4: gt-udp-tun0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP,80000> mtu 1500 state UP qlen 100
    inet6 fe80::bbb1:fd6a:e0ba:5653/64 scope link stable-privacy
       valid_lft forever preferred_lft forever
5: dsvpn0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP,80000> mtu 9000 state UNKNOWN qlen 500
    inet6 64:ff9b::aff:fb01 peer 64:ff9b::aff:fb02/96 scope global
       valid_lft forever preferred_lft forever
    inet6 fe80::c558:1070:2572:534a/64 scope link stable-privacy
       valid_lft forever preferred_lft forever
6: omr-bonding: <BROADCAST,MULTICAST,MASTER,UP,LOWER_UP,80000> mtu 1440 state UP qlen 1000
    inet6 fe80::5030:13ff:fe32:404/64 scope link
       valid_lft forever preferred_lft forever
15: tun0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP,80000> mtu 1500 state UNKNOWN qlen 100
    inet6 fe80::46ea:fcc3:dbb7:7689/64 scope link stable-privacy
       valid_lft forever preferred_lft forever
16: mlvpn0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP,80000> mtu 1444 state UNKNOWN qlen 500
    inet6 fe80::e8cb:1d6c:e6ba:27c6/64 scope link stable-privacy
       valid_lft forever preferred_lft forever
19: tun1: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP,80000> mtu 1500 state UNKNOWN qlen 100
    inet6 fe80::4080:c5e5:ffd0:aa8e/64 scope link stable-privacy
       valid_lft forever preferred_lft forever
54: gt-tun0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP,80000> mtu 1500 state UNKNOWN qlen 500
    inet6 fe80::6041:3e39:7a89:bb7b/64 scope link stable-privacy
       valid_lft forever preferred_lft forever

Ysurac, 2023-07-09:
The problem seems to be in ```/etc/openmptcprouter-vps-admin/omr-6in4/user0```, the "REMOTEIP6" part. What do you have in this file ?

qweryu342, 2023-07-09:
nothing, only "wget -O - https://www.openmptcprouter.com/server/debian-x86_64.sh | sh"

also tried to disable ipv6 and reinstall. not helped:
sudo nano /etc/sysctl.conf

net.ipv6.conf.all.disable_ipv6=1
net.ipv6.conf.default.disable_ipv6=1
net.ipv6.conf.lo.disable_ipv6 = 1

Ysurac, 2023-07-09:
What is the VPN used ? (default is glorytun-tcp)

qweryu342, 2023-07-09:
It is install from scratch. Not  changing any setting. all default

omr6in4 not starting also

root@p647538:/etc/openmptcprouter-vps-admin# systemctl status omr6in4@user0.service
● omr6in4@user0.service - OMR6IN4 on user0
     Loaded: loaded (/lib/systemd/system/omr6in4@.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Sun 2023-07-09 20:51:53 MSK; 5s ago
    Process: 44606 ExecStart=/usr/local/bin/omr-6in4-run start /etc/openmptcprouter-vps-admin/omr-6in4/user0 (co>
   Main PID: 44606 (code=exited, status=1/FAILURE)
        CPU: 2ms

Jul 09 20:51:53 p647538.kvmvps systemd[1]: Starting OMR6IN4 on user0...
Jul 09 20:51:53 p647538.kvmvps omr-6in4-run[44606]: usage: omr-6in4-run start FILE
Jul 09 20:51:53 p647538.kvmvps systemd[1]: omr6in4@user0.service: Main process exited, code=exited, status=1/FAI>
Jul 09 20:51:53 p647538.kvmvps systemd[1]: omr6in4@user0.service: Failed with result 'exit-code'.
Jul 09 20:51:53 p647538.kvmvps systemd[1]: Failed to start OMR6IN4 on user0.

vempire-ghost, 2023-07-10:
I'm experiencing the same issue on Google Cloud. The OMR6IN4 fails to start, and the OMR Service keeps restarting the OMR Service admin every second.
vempire@instance-1:~$ systemctl status omr.service
● omr.service - OMR
     Loaded: loaded (/lib/systemd/system/omr.service; enabled; vendor preset: enabled)
     Active: active (running) since Sun 2023-07-09 22:51:28 -03; 37min ago
   Main PID: 597 (omr-service)
      Tasks: 6 (limit: 4695)
     Memory: 3.9M
        CPU: 2min 57.879s
     CGroup: /system.slice/omr.service
             ├─  597 /bin/bash /usr/local/bin/omr-service
             ├─48631 /bin/bash /usr/local/bin/omr-service
             ├─48632 /bin/bash /usr/local/bin/omr-service
             ├─48634 /bin/bash /usr/local/bin/omr-service
             └─48635 /bin/bash /usr/local/bin/omr-service

Jul 09 23:25:14 instance-1 OMR-Service[43737]: Restart OMR-Admin
Jul 09 23:25:32 instance-1 OMR-Service[44120]: Restart OMR-Admin
Jul 09 23:26:08 instance-1 OMR-Service[44880]: Restart OMR-Admin
Jul 09 23:26:26 instance-1 OMR-Service[45259]: Restart OMR-Admin
Jul 09 23:26:44 instance-1 OMR-Service[45639]: Restart OMR-Admin
Jul 09 23:27:21 instance-1 OMR-Service[46397]: Restart OMR-Admin
Jul 09 23:27:39 instance-1 OMR-Service[46776]: Restart OMR-Admin
Jul 09 23:27:58 instance-1 OMR-Service[47155]: Restart OMR-Admin
Jul 09 23:28:36 instance-1 OMR-Service[47913]: Restart OMR-Admin
Jul 09 23:28:54 instance-1 OMR-Service[48292]: Restart OMR-Admin

vempire@instance-1:~$ systemctl status omr6in4@user0.service
● omr6in4@user0.service - OMR6IN4 on user0
     Loaded: loaded (/lib/systemd/system/omr6in4@.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Sun 2023-07-09 22:51:29 -03; 40min ago
    Process: 603 ExecStart=/usr/local/bin/omr-6in4-run start /etc/openmptcprouter-vps-admin/omr-6in4/user0 (code=exited, status=1/FAILURE)
   Main PID: 603 (code=exited, status=1/FAILURE)
        CPU: 4ms

Jul 09 22:51:28 instance-1 systemd[1]: Starting OMR6IN4 on user0...
Jul 09 22:51:28 instance-1 omr-6in4-run[603]: usage: omr-6in4-run start FILE
Jul 09 22:51:29 instance-1 systemd[1]: omr6in4@user0.service: Main process exited, code=exited, status=1/FAILURE
Jul 09 22:51:29 instance-1 systemd[1]: omr6in4@user0.service: Failed with result 'exit-code'.
Jul 09 22:51:29 instance-1 systemd[1]: Failed to start OMR6IN4 on user0.

Ysurac, 2023-07-10:
What is the result of ```uname -a```, ```python3 -V```, ```systemd-detect-virt`` and ```cat /etc/os-release``` ?

vempire-ghost, 2023-07-10:
root@instance-1:/home/vempire# uname -a
Linux instance-1 5.4.207-mptcp #1 SMP Sun Jul 24 14:39:44 UTC 2022 x86_64 GNU/Linux

root@instance-1:/home/vempire# python3 -V
Python 3.9.2

root@instance-1:/home/vempire# systemd-detect-virt
kvm

root@instance-1:/home/vempire# cat /etc/os-release
PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"

duckigol, 2023-07-10:
Manually Shadowsocks and glorytun does work but omr-admin doesn't, I decided to stop it instead. systemctl stop omr-admin.

qweryu342, 2023-07-10:
root@217.144.98.99:65222:~# uname -a
Linux p647538.kvmvps 5.4.207-mptcp #1 SMP Sun Jul 24 14:39:44 UTC 2022 x86_64 GNU/Linux

root@217.144.98.99:65222:~# python3 -V
Python 3.9.2

root@217.144.98.99:65222:~# systemd-detect-virt
kvm

root@217.144.98.99:65222:~# cat /etc/os-release
PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"

docblocxxx, 2023-07-10:
Seems I'm not alone, I'm having the same problem.
Clean Debian 11 x64
and the result is like everyone else.
python gives error error: look-around, including look-ahead and look-behind, is not supported

.py doesn't work correctly.

docblocxxx, 2023-07-10:
And as a result, it is impossible to get the current config for shadowsocks VPS, it is impossible to get the *status* and other things.

xxzj2012, 2023-07-11:
I have  same problem，hao to solve

aliosa27, 2023-07-11:
looks like the ipv6 check was disabled for now from the script, you should be able to rerun the vps script to pull the latest omr-admin.py

Ysurac, 2023-07-11:
Can you try to install VPS script again ? ```wget -O - https://www.openmptcprouter.com/server/debian-x86_64.sh | sh```

karelfridrich, 2023-07-11:
@Ysurac Latest script solved the problem for me on Ubuntu 20.04. Thanks!

closed 2023-07-14 by qweryu342