Ysurac/openmptcprouter #2879: Support for RAID Controllers

marius137, 2023-07-15:
When I tried running OMR on an HP DL380G6 with both the integrated HP P410 raid controller and a random Adaptec Raid controller the boot hangs at `waiting for root device PARTUUID=`.

Ysurac, 2023-07-15:
What is the version tested ?

marius137, 2023-07-15:
I tried the ext4 non-uefi version downloaded today

[openmptcprouter-v0.59.1-5.4-r0+16594-ce92de8c8c-x86-64-generic-ext4-combined.img.gz](https://download.openmptcprouter.com/release/v0.59.1/x86_64/targets/x86/64/openmptcprouter-v0.59.1-5.4-r0+16594-ce92de8c8c-x86-64-generic-ext4-combined.img.gz)

Ysurac, 2023-07-16:
Can you try snapshot image ? https://snapshots.openmptcprouter.com/6.1/x86_64/targets/x86/64/
If this is still not working, do you know needed modules ?

marius137, 2023-07-19:
sorry for the late reply, I couldn't try the snapshot as I needed the hardware, but I did some more digging and it seems that openwrt in general doesn't play nice with raid controllers. I ended up just using the normal sata ports on the mainboard and a single drive without raid.

I am closing the issue because this probably is an openwrt issue, and not really OMR related

closed 2023-07-19 by marius137