Ysurac/openmptcprouter #2880: Support ECC Memory

Kalimeiro, 2023-07-15:
Hi, we have servers with ecc memory and memory working in non-ecc mode. it's possible to enable them for intel/amd ?

Actually, i can't see EDAC loaded in dmesg and "dmidecode --type memory" show None in "Error Correction Type" instead "Multi-bit ECC"

Thanks.

Kalimeiro, 2023-07-18:
I use only 5.4 kernel because on 6.1 agregation not working.

Ysurac, 2023-07-18:
Aggregation is working on 6.1, 5.4 kernel will be put as legacy/deprecated.

Kalimeiro, 2023-07-19:
Je suis désolé de te contredire, mais l'agrégation avec le noyau 6.1 ne fonctionne pas du tout, il n'y a que l'interface MASTER qui est utilisé.

De plus, la redirection de port ne fonctionne plus et je suis en train de chercher quel est le problème depuis hier soir.

Kalimeiro, 2023-07-19:
ok i'm testing 6.1 kernel and agregation is working now, be careful to set mptcp version 1 and not version 0 when switch in 6.1 kernel.

the latest snapshot not integrate edac_mc module.

closed 2023-07-23 by Kalimeiro