Ysurac/openmptcprouter #2881: Compilation of custom rpi4 image failed

ByteArrayException, 2023-07-16:
I want to build an custom image for rpi4 on an Ubuntu 20.04.6 LTS.
I run the following command in cloned Repo Dir: 
OMR_TARGET="rpi4" OMR_FEED_SRC="develop" ./build.sh V=s

I received many errors like: 
Installing package 'syslogd' from openmptcprouter
WARNING: No feed for package 'libc' found
WARNING: No feed for package 'librt' found
WARNING: No feed for package 'libpthread' found
WARNING: No feed for package 'busybox' found
Installing package 'openmptcprouter-mini' from openmptcprouter
WARNING: No feed for package 'libc' found
WARNING: No feed for package 'librt' found
WARNING: No feed for package 'libpthread' found
WARNING: No feed for package 'f2fs-tools' found
WARNING: No feed for package 'dnsmasq-full' found
WARNING: No feed for package 'uhttpd' found
WARNING: No feed for package 'uhttpd-mod-ubus' found
WARNING: No feed for package 'arptables' found
WARNING: No feed for package 'libnetfilter-conntrack' found
WARNING: No feed for package 'iptables-mod-hashlimit' found
WARNING: No feed for package 'iptables-mod-trace' found
WARNING: No feed for package 'iptables-mod-conntrack-extra' found
WARNING: No feed for package 'kmod-nf-nat' found
WARNING: No feed for package 'kmod-nf-nathelper' found
WARNING: No feed for package 'kmod-nf-nathelper-extra' found
WARNING: No feed for package 'iptables-mod-extra' found
WARNING: No feed for package 'iptables-mod-ipsec' found
WARNING: No feed for package 'kmod-crypto-authenc' found
WARNING: No feed for package 'kmod-ipsec' found
WARNING: No feed for package 'kmod-ipsec4' found
WARNING: No feed for package 'kmod-ipsec6' found
WARNING: No feed for package 'kmod-ipt-ipsec' found
WARNING: No feed for package 'wireless-tools' found
WARNING: No feed for package 'libiwinfo-lua' found
WARNING: No feed for package 'ca-bundle' found
WARNING: No feed for package 'ca-certificates' found
WARNING: No feed for package 'libustream-openssl' found
Installing package 'luci-app-sqm' from luci
WARNING: No feed for source package 'lua' found
WARNING: No feed for package 'libc' found
WARNING: No feed for package 'librt' found
WARNING: No feed for package 'libpthread' found
WARNING: No feed for package 'libc' found
WARNING: No feed for package 'librt' found
WARNING: No feed for package 'libpthread' found
WARNING: No feed for package 'libc' found
WARNING: No feed for package 'librt' found
WARNING: No feed for package 'libpthread' found
WARNING: No feed for package 'libc' found
WARNING: No feed for package 'librt' found
WARNING: No feed for package 'libpthread' found
WARNING: No feed for package 'libc' found
WARNING: No feed for package 'librt' found
WARNING: No feed for package 'libpthread' found
WARNING: No feed for package 'libc' found
WARNING: No feed for package 'librt' found
WARNING: No feed for package 'libpthread' found
WARNING: No feed for package 'libc' found
WARNING: No feed for package 'librt' found
WARNING: No feed for package 'libpthread' found
WARNING: No feed for package 'libc' found
WARNING: No feed for package 'librt' found
WARNING: No feed for package 'libpthread' found
WARNING: No feed for package 'libc' found
WARNING: No feed for package 'librt' found
WARNING: No feed for package 'libpthread' found
WARNING: No feed for package 'libc' found
WARNING: No feed for package 'librt' found
WARNING: No feed for package 'libpthread' found
[... 131 lines cut ...]
WARNING: No feed for package 'kmod-usb-net-huawei-cdc-ncm' found
WARNING: No feed for package 'kmod-usb-net-rndis' found
WARNING: No feed for package 'kmod-usb-net-cdc-ether' found
WARNING: No feed for package 'kmod-usb-net-ipheth' found
WARNING: No feed for package 'kmod-rt2800-usb' found
WARNING: No feed for package 'kmod-rtl8xxxu' found
WARNING: No feed for package 'kmod-rtl8192cu' found
WARNING: No feed for package 'kmod-net-rtl8192su' found
WARNING: No feed for package 'wpad-basic' found
WARNING: No feed for package 'kmod-mt7601u' found
WARNING: No feed for package 'kmod-rtl8187' found
WARNING: No feed for package '464xlat' found
WARNING: No feed for package 'kmod-zram' found
WARNING: No feed for package 'kmod-swconfig' found
WARNING: No feed for package 'swconfig' found
WARNING: No feed for package 'kmod-ipt-nat' found
WARNING: No feed for package 'kmod-ipt-nat6' found
WARNING: No feed for package 'kmod-tcp-nanqinlang' found
WARNING: No feed for package 'iptables-mod-ipopt' found
WARNING: No feed for package 'block-mount' found
WARNING: No feed for package 'blockd' found
WARNING: No feed for package 'fstools' found

Is that an known issue?
Many thanks already for your help!

Ysurac, 2023-07-16:
What is the error ?

ByteArrayException, 2023-07-16:
Okay, I correct myself, no error but warnings, but this is not normal that it does not find any packages and at the end comes the toplevel.mk line 180 error:
Prerequisite check failed. Use FORCE=1 to override.
make[1]: *** [/root/OMRBuild/rpi4/5.4/source/include/toplevel.mk:180: staging_dir/host/.prereq-build] Error 1
make: *** [/root/OMRBuild/rpi4/5.4/source/include/toplevel.mk:77: prepare-tmpinfo] Error 2
Cannot open './tmp/.packageinfo': No such file or directory
Can't open file './tmp/.targetinfo': No such file or directory
Installing package 'macremapper' from packages
WARNING: No feed for package 'kmod-cfg80211' found
WARNING: No feed for package 'kmod-br-netfilter' found
Done
Building openmptcprouter for the target rpi4 with kernel 5.4
Prerequisite check failed. Use FORCE=1 to override.
make: *** [/root/OMRBuild/rpi4/5.4/source/include/toplevel.mk:180: staging_dir/host/.prereq-build] Error 1

Ysurac, 2023-07-16:
It's develop branch and warning are not a real problem.
But the error is a problem, I think the error is above log you give and seems to be related to a missing package on host.

ByteArrayException, 2023-07-16:
Exactly, that's what I thought too, I used the command from the wiki "apt-get install busybox curl rsync build-essential asciidoc binutils bzip2 gawk gettext git libncurses5-dev libz-dev patch unzip zlib1g-dev lib32gcc1 libc6-dev-i386 subversion flex uglifyjs git-core gcc-multilib p7zip p7zip-full msmtp libssl-dev texinfo libglib2. 0-dev xmlto qemu-utils upx libelf-dev autoconf automake libtool autopoint device-tree-compiler wget tar"
and the command from the GitHub workflow
"apt install build-essential asciidoc binutils bzip2 gawk gettext git libncurses5-dev libz-dev patch unzip zlib1g-dev lib32gcc-s1 libc6-dev-i386 subversion flex uglifyjs git-core gcc-multilib p7zip p7zip-full msmtp libssl-dev texinfo libglib2. 0-dev xmlto qemu-utils upx libelf-dev autoconf automake libtool autopoint device-tree-compiler python3-pyelftools" has been run before and this error still appears.
Furthermore, the setup says it cannot find mkhash.

Ysurac, 2023-07-18:
Automatic compilation is done via Github using Ubuntu, so this should work...
Check that you have enough disk space, and I still need some log before the error.

ByteArrayException, 2023-07-19:
I'd be happy to submit that. Here as a post or rather via pastebin dump? :D

closed 2023-10-24 by github-actions[bot]