Ysurac/openmptcprouter #2883: NTP Time issue - device behind OMR not connecting

benw8080, 2023-07-18:
## Expected Behavior
I have a TIMEMachines NTP clock, I expect it to connect correctly to pool.ntp.org (oceania.pool.ntp.org)

## Current Behavior
TimeMachines POE NTP Clock, currently NOT connecting to NTP time server
Device is poe powered, 
OMR Router --> Ubiquity DreamRouter --> POE to clock 
(poe is in spec, all cables tested and working)

On another device on same network (same setup/wireless), I can ping by IP and DNS to pool.ntp.org
OMR is connected to internet using 2 x 4G modems on seperate ethernet ports, 
both WAN networks are set to DHCP mode
Both 4G routers are set to different Ip address's (3rd oct different)
Ubiquity DRouter (uDR) is set to a different IP range, no conflict there.

if I remove OMR from path and connect uDR directly to a 4G modem, the clock will receive the correct update/connection to the time server.

Thanks for any and all help with this!

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.59.1-5.4 r0+16594-ce92de
  - OpenMPTCProuter VPS version: 0.1026 5.4.100 mptcp
  - OpenMPTCProuter VPS provider: Mamoth (linux QLD Aus)
  - OpenMPTCProuter platform: x86 - Ubuntu - vps

benw8080, 2023-07-24:
> VPN is up in System->OpenMPTCProuter, status page ?
> Can you update VPS time using same pool from the VPS ?
> Can you update time using NTP from another computer on same network ?

Hey Ysurac,

confirmed VPN running in Status page.
confirmed can connect to ntp pool server from other device (

possible resolution:
I used ping from terminal on OMR Router to oceania.pool.ntp.org <http://oceania.pool.ntp.org/>, this appears to have resolved the issue as the Clock(s) can now see the NTP pool server.

I’m unsure why this would/could resolve it, (It may not have anything to do with resolving the connection).

I did add a FW rule to allow :123 to all LAN (I’m not certain this is the reason things work now, though when I disable it - there seems to be a negative affect)

I updated the settings in OMR to sync with NTP time server, it was previously on the default settings.
- time updated correctly
- NTp client and server check box ‘ticked'
- added oceania.pool.ntp.org <http://oceania.pool.ntp.org/> to the OMR NTP server list

 Q: Does “Allow DHCP advertised servers” mean OMR Host can be used as NTP Server? - currently this is not checked.

- adjusted server refresh on clock(s) to 2min, default 10 min.

Thanks for your help,

ben

Ysurac, 2023-07-24:
You shouldn't have to open ports if you are on same subnet as OpenMPTCProuter router.
To use OpenMPTCProuter as NTP server, it's in menu System->System, "Time synchronization" tab:
[image]

benw8080, 2023-07-26:
Hi Ysurac,

use as server was selected (I believe this is defualt), the version I had was using 0.openwrt.pool.ntp.org <http://0.openwrt.pool.ntp.org/>.
After replacing the pool server info with ociania.pool.ntp.org <http://ociania.pool.ntp.org/> and pool.ntp.org <http://pool.ntp.org/> things seem to be more stable, Ive also deleted the port forward and reconfigured the internal clocks to use OMR as the default NTP server.

Thanks for your help, this has been a great result.

Cheers,

ben

—————————

closed 2023-10-30 by github-actions[bot]