Ysurac/openmptcprouter #2884: Vps configuration for ipv6 use

brazzo97, 2023-07-18:
Hi i am trying to configure omr using a vps selfhosted in a location where i have a fiber  optic connection , i did ipv4 and all works great but when i try to configure ipv6 i cant get it to work .

on the vps i set :   

iface ens32 inet6 dhcp ( that gets me an ipv6 ) 
 request_prefix 1 ( that should request a prefix from the dhcpv6 server ) 

infact doing ip -a in the vps says that i got an ipv6 /128 and   pinging google.it using v6 works.
if i check on the router it says that it has assigned an ipv6 prefix /56 to the ipv6 of the vps so i think it is correct .

after that i installed omr and once i rebooted i cant no longer ping v6 on the vps , its strange because the ipv6 is still assigned to the eth interface , i can ping the gateway but i cant ping outside and the route is still there .

How can i fix , and then what are the step to have ipv6 working on omr client ? Thanks

Ysurac, 2023-07-18:
Please use the template when you open an issue.
I would need the script you used (stable or snapshot VPS script ?)
What do you have in ```/etc/shorewall6/params.net``` and the result of ```ip -6 a```.

brazzo97, 2023-07-20:
Hi i used the stable vps script , 

here is the ip -6 a 

root@vps:~# ip -6 a
1: lo: <LOOPBACK,UP,LOWER_UP,80000> mtu 65536 state UNKNOWN qlen 1000
    inet6 ::1/128 scope host
       valid_lft forever preferred_lft forever
2: ens32: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 state UP qlen 1000
    inet6 2a0d:xxxx:xxxx:xxxx::1/56 scope global
       valid_lft forever preferred_lft forever
    inet6 fe80::250:56ff:fe36:260c/64 scope link
       valid_lft forever preferred_lft forever
4: dsvpn0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP,80000> mtu 9000 state UNKNOWN qlen 500
    inet6 64:ff9b::aff:fb01 peer 64:ff9b::aff:fb02/96 scope global
       valid_lft forever preferred_lft forever
    inet6 fe80::104a:98f:930:4922/64 scope link stable-privacy
       valid_lft forever preferred_lft forever
5: gt-udp-tun0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP,80000> mtu 1500 state UP qlen 100
    inet6 fe80::d13c:b812:87cc:832f/64 scope link stable-privacy
       valid_lft forever preferred_lft forever
7: omr-bonding: <BROADCAST,MULTICAST,MASTER,UP,LOWER_UP,80000> mtu 1440 state UP qlen 1000
    inet6 fe80::5c86:17ff:fe1d:3f24/64 scope link
       valid_lft forever preferred_lft forever
13: tun1: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP,80000> mtu 1500 state UNKNOWN qlen 500
    inet6 fe80::c880:5adf:d8c1:3c52/64 scope link stable-privacy
       valid_lft forever preferred_lft forever
15: tun0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP,80000> mtu 1500 state UNKNOWN qlen 500
    inet6 fe80::8a0:830e:a881:e9dd/64 scope link stable-privacy
       valid_lft forever preferred_lft forever
17: mlvpn0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP,80000> mtu 1444 state UNKNOWN qlen 500
    inet6 fe80::1c40:6c34:955e:d139/64 scope link stable-privacy
       valid_lft forever preferred_lft forever
65: omr-6in4-user0@NONE: <POINTOPOINT,NOARP,UP,LOWER_UP,80000> mtu 1480 state UNKNOWN qlen 1000
    inet6 fe80::aff:ff01/64 scope link
       valid_lft forever preferred_lft forever
    inet6 fe80::a00:1/126 scope link
       valid_lft forever preferred_lft forever
67: gt-tun0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP,80000> mtu 1500 state UNKNOWN qlen 500
    inet6 fe80::c43:f2b1:6846:77bf/64 scope link stable-privacy
       valid_lft forever preferred_lft forever
[image]

Ysurac, 2023-07-20:
All seems good here.
What is the result of ```ip -6 r```  and ```ip6tables-save``` ?

brazzo97, 2023-07-20:
root@vps:~# ip -6 r
::1 dev lo proto kernel metric 256 pref medium
2a0d:xxxx:xxxx:xxxx::/56 dev ens32 proto kernel metric 256 pref medium
2a0d:yyyy:yyyy:yyyy::/56 via fe80::a00:2 dev omr-6in4-user0 metric 1024 pref medium
2a0d:xxxx:xxxx::/48 dev ens32 proto kernel metric 256 expires 86314sec pref medium
fe80::a00:0/126 dev omr-6in4-user0 proto kernel metric 256 pref medium
fe80::/64 dev ens32 proto kernel metric 256 pref medium
fe80::/64 dev dsvpn0 proto kernel metric 256 pref medium
fe80::/64 dev tun0 proto kernel metric 256 pref medium
fe80::/64 dev tun1 proto kernel metric 256 pref medium
fe80::/64 dev gt-udp-tun0 proto kernel metric 256 pref medium
fe80::/64 dev mlvpn0 proto kernel metric 256 pref medium
fe80::/64 dev omr-bonding proto kernel metric 256 pref medium
fe80::/64 dev omr-6in4-user0 proto kernel metric 256 pref medium
fe80::/64 dev gt-tun0 proto kernel metric 256 pref medium
default via 2a0d:xxxx:xxxx:: dev ens32 metric 1024 onlink pref medium
default via fe80::20c:29ff:fe1c:bdcc dev ens32 proto ra metric 1024 expires 1714sec hoplimit 64 pref high

# Generated by ip6tables-save v1.8.7 on Thu Jul 20 11:38:01 2023
*mangle
:PREROUTING ACCEPT [627474:1167267013]
:INPUT ACCEPT [615764:1164778043]
:FORWARD ACCEPT [11629:2482778]
:OUTPUT ACCEPT [516454:60698832]
:POSTROUTING ACCEPT [527957:63171530]
:rpfilter - [0:0]
:rplog - [0:0]
:tcfor - [0:0]
:tcin - [0:0]
:tcout - [0:0]
:tcpost - [0:0]
:tcpre - [0:0]
-A PREROUTING -j tcpre
-A PREROUTING -i ens32 -j rpfilter
-A INPUT -j tcin
-A FORWARD -j MARK --set-xmark 0x0/0xff
-A FORWARD -j tcfor
-A OUTPUT -j tcout
-A POSTROUTING -j tcpost
-A rpfilter -m rpfilter --validmark --invert -m conntrack --ctstate INVALID,NEW,RELATED -j rplog
-A rplog -j LOG --log-prefix "Shorewall:rplog:DROP:" --log-level 6
-A rplog -j DROP
COMMIT
# Completed on Thu Jul 20 11:38:01 2023
# Generated by ip6tables-save v1.8.7 on Thu Jul 20 11:38:01 2023
*raw
:PREROUTING ACCEPT [627475:1167267062]
:OUTPUT ACCEPT [516455:60698881]
-A PREROUTING -p udp -m udp --dport 10080 -j CT --helper amanda
-A PREROUTING -p tcp -m tcp --dport 21 --tcp-flags FIN,SYN,RST,ACK SYN -j CT --helper ftp
-A PREROUTING -p udp -m udp --dport 1719 -j CT --helper RAS
-A PREROUTING -p tcp -m tcp --dport 1720 --tcp-flags FIN,SYN,RST,ACK SYN -j CT --helper Q.931
-A PREROUTING -p tcp -m tcp --dport 6566 --tcp-flags FIN,SYN,RST,ACK SYN -j CT --helper sane
-A PREROUTING -p udp -m udp --dport 5060 -j CT --helper sip
-A PREROUTING -p udp -m udp --dport 69 -j CT --helper tftp
-A OUTPUT -p udp -m udp --dport 10080 -j CT --helper amanda
-A OUTPUT -p tcp -m tcp --dport 21 --tcp-flags FIN,SYN,RST,ACK SYN -j CT --helper ftp
-A OUTPUT -p udp -m udp --dport 1719 -j CT --helper RAS
-A OUTPUT -p tcp -m tcp --dport 1720 --tcp-flags FIN,SYN,RST,ACK SYN -j CT --helper Q.931
-A OUTPUT -p tcp -m tcp --dport 6566 --tcp-flags FIN,SYN,RST,ACK SYN -j CT --helper sane
-A OUTPUT -p udp -m udp --dport 5060 -j CT --helper sip
-A OUTPUT -p udp -m udp --dport 69 -j CT --helper tftp
COMMIT
# Completed on Thu Jul 20 11:38:01 2023
# Generated by ip6tables-save v1.8.7 on Thu Jul 20 11:38:01 2023
*filter
:INPUT DROP [0:0]
:FORWARD DROP [0:0]
:OUTPUT DROP [0:0]
:AllowICMPs - [0:0]
[... 114 lines cut ...]
-A vpn-fw -m conntrack --ctstate RELATED,ESTABLISHED -j ACCEPT
-A vpn-fw -p ipv6-icmp -j ACCEPT
-A vpn-fw -j ACCEPT
-A vpn-net -m conntrack --ctstate RELATED,ESTABLISHED -j ACCEPT
-A vpn-net -p ipv6-icmp -j ACCEPT
-A vpn-net -p udp -m udp --dport 53 -j ACCEPT
-A vpn-net -j ACCEPT
-A vpn_frwd -m conntrack --ctstate INVALID,NEW,UNTRACKED -j dynamic
-A vpn_frwd -p tcp -j tcpflags
-A vpn_frwd -o ens32 -j vpn-net
-A vpn_frwd -o omr-6in4-user+ -j ACCEPT
COMMIT
# Completed on Thu Jul 20 11:38:01 2023
# Generated by ip6tables-save v1.8.7 on Thu Jul 20 11:38:01 2023
*nat
:PREROUTING ACCEPT [5237:535121]
:INPUT ACCEPT [8:640]
:OUTPUT ACCEPT [4231:389500]
:POSTROUTING ACCEPT [9170:892528]
:ens32_masq - [0:0]
-A POSTROUTING -o ens32 -j ens32_masq
-A ens32_masq -s fe80::/10 -j MASQUERADE
-A ens32_masq -s fd00::/8 -j MASQUERADE
COMMIT
# Completed on Thu Jul 20 11:38:01 2023

brazzo97, 2023-07-20:
when i enable ipv6 in the omr client i start using ipv6 in the clients but its natted on the vps ipv6 if i put the prefix subnet i have in the box under the enable ipv6 after a save it goes back to stock local subnet , i had to go to network/interface/global and put the prefix here for the router to start giving those ip to client , but still if i look at what ip i am using with a service like myip i see the vps ipv6 not the ip omr leased to my client

Ysurac, 2023-07-20:
I don't really understand what you mean, it's working ? 
Output IP is always VPS IP due to proxy use (it's possible to output using different IPs with multiple Shadowsocks servers but I think I forgot to write a doc or maybe not yet implemented...).

brazzo97, 2023-07-20:
wait i want to use pd to give each user an ipv6 address from my /56 subnet , omr dhcp server gives client an ipv6 from that subnet but you connect to internet using the vps ipv6 not the public ipv6 from the /56 pd subnet

Ysurac, 2023-07-20:
With the proxy enabled yes.

sp2003, 2023-07-26:
so it's impossible to use a public prefix with omr?
Then Why there is the option under IPv6 enable ?

closed 2023-10-30 by github-actions[bot]